balenasolutions/browser #23: Browser crashes when trying to serve local video

sradevski, 2020-09-04:
I created a very simple server that serves a static mp4 file, and whenever the browser tries to read it it crashes with:

```
[Logs]    [9/4/2020, 2:51:07 PM] [browser] hostname: No address associated with hostname
[Logs]    [9/4/2020, 2:51:07 PM] [browser] xauth:  file /home/chromium/.Xauthority does not exist
[Logs]    [9/4/2020, 2:51:07 PM] [browser] xauth: (stdin):1:  bad display name "aff8c20:0" in "add" command
[Logs]    [9/4/2020, 2:51:07 PM] [browser] 
[Logs]    [9/4/2020, 2:51:07 PM] [browser] 
[Logs]    [9/4/2020, 2:51:07 PM] [browser] X.Org X Server 1.20.4
[Logs]    [9/4/2020, 2:51:07 PM] [browser] X Protocol Version 11, Revision 0
[Logs]    [9/4/2020, 2:51:07 PM] [browser] Build Operating System: Linux 4.15.0-99-generic armv8l Raspbian
[Logs]    [9/4/2020, 2:51:07 PM] [browser] Current Operating System: Linux aff8c20 4.19.75 #1 SMP Thu Jun 4 14:39:26 UTC 2020 armv7l
[Logs]    [9/4/2020, 2:51:07 PM] [browser] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=1184 bcm2708_fb.fbheight=624 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:AB:E2:34 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 rootfstype=ext4 rootwait root=UUID=ba1eadef-cae9-446c-a81f-1a39b4c52dab rootwait
[Logs]    [9/4/2020, 2:51:07 PM] [browser] Build Date: 07 July 2020  09:38:51PM
[Logs]    [9/4/2020, 2:51:07 PM] [browser] xorg-server 2:1.20.4-1+rpt2 (https://www.debian.org/support) 
[Logs]    [9/4/2020, 2:51:07 PM] [browser] Current version of pixman: 0.36.0
[Logs]    [9/4/2020, 2:51:07 PM] [browser] 	Before reporting problems, check http://wiki.x.org
[Logs]    [9/4/2020, 2:51:07 PM] [browser] 	to make sure that you have the latest version.
[Logs]    [9/4/2020, 2:51:07 PM] [browser] Markers: (--) probed, (**) from config file, (==) default setting,
[Logs]    [9/4/2020, 2:51:07 PM] [browser] 	(++) from command line, (!!) notice, (II) informational,
[Logs]    [9/4/2020, 2:51:07 PM] [browser] 	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[Logs]    [9/4/2020, 2:51:07 PM] [browser] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Sep  4 12:51:07 2020
[Logs]    [9/4/2020, 2:51:07 PM] [browser] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[Logs]    [9/4/2020, 2:51:08 PM] [browser]  --disable-quic --enable-tcp-fast-open --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0 --ppapi-flash-version=
[Logs]    [9/4/2020, 2:51:10 PM] [browser] [114:154:0904/125110.806190:ERROR:bus.cc(393)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[Logs]    [9/4/2020, 2:51:11 PM] [browser] libEGL warning: DRI2: failed to authenticate
[Logs]    [9/4/2020, 2:51:11 PM] [browser] [114:345:0904/125111.585364:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[Logs]    [9/4/2020, 2:51:11 PM] [browser] [114:345:0904/125111.673111:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[Logs]    [9/4/2020, 2:51:11 PM] [browser] [114:345:0904/125111.673679:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[Logs]    [9/4/2020, 2:51:12 PM] [browser] [114:368:0904/125112.036430:ERROR:bus.cc(393)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[Logs]    [9/4/2020, 2:51:12 PM] [browser] [114:368:0904/125112.037712:ERROR:bus.cc(393)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[Logs]    [9/4/2020, 2:51:12 PM] [browser] [114:368:0904/125112.038577:ERROR:bus.cc(393)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[Logs]    [9/4/2020, 2:51:12 PM] [browser] [114:368:0904/125112.040527:ERROR:bus.cc(393)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[Logs]    [9/4/2020, 2:51:12 PM] [browser] [114:368:0904/125112.041620:ERROR:bus.cc(393)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[Logs]    [9/4/2020, 2:51:12 PM] [browser] [163:163:0904/125112.345495:ERROR:sandbox_linux.cc(369)] InitializeSandbox() called with multiple threads in process gpu-process.
[Logs]    [9/4/2020, 2:51:12 PM] [browser] 
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): Gtk-WARNING **: 12:51:12.793: Could not load a pixbuf from icon theme.
[Logs]    [9/4/2020, 2:51:12 PM] [browser] This may indicate that pixbuf loaders or the mime database could not be found.
[Logs]    [9/4/2020, 2:51:12 PM] [browser] 
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): GdkPixbuf-CRITICAL **: 12:51:12.794: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[Logs]    [9/4/2020, 2:51:12 PM] [browser] 
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): GdkPixbuf-CRITICAL **: 12:51:12.795: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[Logs]    [9/4/2020, 2:51:12 PM] [browser] 
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): GdkPixbuf-CRITICAL **: 12:51:12.800: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): GdkPixbuf-CRITICAL **: 12:51:12.800: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[Logs]    [9/4/2020, 2:51:12 PM] [browser] 
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): GdkPixbuf-CRITICAL **: 12:51:12.804: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[Logs]    [9/4/2020, 2:51:12 PM] [browser] 
[Logs]    [9/4/2020, 2:51:12 PM] [browser] (chromium-browser-v7:114): GdkPixbuf-CRITICAL **: 12:51:12.804: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[Logs]    [9/4/2020, 2:51:13 PM] [browser] usrvcsm: [vcsm_malloc_cache]: [163] [ChromeBuf]: ioctl mem-alloc FAILED [-1] (hdl: 0)

```

See this repo: https://github.com/balena-io-playground/steve-play-with-blocks

phil-d-wilson, 2020-09-08:
The logging here are not actually errors, they are verbose messages which you are seeing because you have `DEBUG: 1` in your compose file. We are going to remove this variable, since I think there is more noise and more potential for misunderstanding with people using this block, than actual benefit to the logging output itself.

What the issue is here, after recreating, is the browser block does not work when in local mode. This needs to be investigated.

phil-d-wilson, 2021-04-06:
Fixed in version 2

closed 2021-04-06 by phil-d-wilson