balenasolutions/browser #24: Browser needs to reconnect to local server if the local server container restarts

sradevski, 2020-09-04:
I found it pretty handy that the browser tries to look for a local service running on port 80, but the functionality is pretty limited. This can still be an opt-in functionality, but what is more practical is if the server itself announces to the browser what endpoint to hit and when, which should give the control for retries, failure handling, etc. to the other services.

I was developing a couple of services locally and using livepush the server would restart, and the browser wouldn't show an empty page and it wouldn't try to reconnect to the server. I can imagine the same thing happening in production and the only way to recover will be to restart the browser service.

rohansingh, 2020-12-31:
This is a major hack but worked for my use case. It refreshes the browser once per second:

```Dockerfile
 FROM balenablocks/browser:raspberrypi4-64

 RUN install_packages xdotool

 CMD ["export DISPLAY=:0 XAUTHORITY=/home/chromium/.Xauthority && (watch -n1 xdotool key ctrl+F5 &)"]
```

phil-d-wilson, 2021-04-06:
Containers can now use the browser block API to tell it to refresh. This solves this issue.

closed 2021-04-06 by phil-d-wilson