balenasolutions/browser #26: Keep command-line option for persistence seperate from FLAGS

rahul-thakoor, 2020-09-08:
https://github.com/balenablocks/browser/blob/a53ab4ce87d08a0ba14eb51a1d7f6640a8618d3e/start.sh#L71

Someone trying to keep persistence working but only modifying some other flags might have to remember to add the above option to the flags?

maybe we export ` PERSISTENT_FLAG="--user-data-dir=/data"` if we detect the variable then do 
`echo "chromium-browser $CHROME_LAUNCH_URL $FLAGS  $PERSISTENT_FLAG   --window-size=$WINDOW_SIZE --window-position=$WINDOW_POSITION $OUTPUT" >> /home/chromium/xstart.sh` here
https://github.com/balenablocks/browser/blob/a53ab4ce87d08a0ba14eb51a1d7f6640a8618d3e/start.sh#L150

rahul-thakoor, 2020-09-08:
@phil-d-wilson I can work on this but would be grateful to get your feedback first

phil-d-wilson, 2020-09-08:
@rahul-thakoor - yes that makes sense. Do you think we should roll this into the other enVar separation work and use this method for all of the options?!?

rahul-thakoor, 2020-09-08:
yes 👍

phil-d-wilson, 2021-04-06:
Fixed in version 2

closed 2021-04-06 by phil-d-wilson