balenasolutions/browser #28: Provide an on-screen keyboard

phil-d-wilson, 2020-09-09:


nated0g, 2021-07-24:
Also interested in this.  I'm trying to get something to work with matchbox-keyboard, but so far no luck.  If I get something working I'll make a PR.

nated0g, 2021-07-24:
@phil-d-wilson Have you made any progress towards this?  I will keep trying myself, but if you've made any steps towards it, it would be great not to be duplicating work :)

orchydea, 2021-12-06:
Hi! Any progress on this @nated0g  @phil-d-wilson?

deregtd, 2022-08-07:
I tried switching to the WPE browser block to get an on screen keyboard, but that block was so buggy I had to go back to this one.  Hoping someone adds a simple OSK at some point here.

DanCarbo, 2023-03-22:
+1
It would be nice to have this feature.

honeytung, 2024-11-05:
Hi! Any updates on the on-screen keyboard? @phil-d-wilson

phil-d-wilson, 2024-11-05:
No plans to work on this @honeytung but I'd help anyone trying to contribute it.

jonmuell, 2024-11-19:
We solved the on-screen keyboard issue by adding a keyboard component to our webapp and extending the Balena browser with an API that takes requests for each keystroke and sends it over to _xdotool_. Unfortunately, this is only working if you have a kiosk like application and doesn't work for browsing the web.

To get the keyboard working everywhere, we eventually switched to Electron. The keyboard now is a separate window that we trigger when a user clicks an input field. For this, we had to inject custom JS to add a listener. It won't work on all sites as some use non-standard inputs but in most cases it will work.

nordson-sonti, 2025-01-28:
> We solved the on-screen keyboard issue by adding a keyboard component to our webapp and extending the Balena browser with an API that takes requests for each keystroke and sends it over to [...]

Hi @jonmuell Could please point me to electron solution so that i can try it out.

srjchauhan, 2025-07-12:
Hey @phil-d-wilson I am also interested in this feature as i need this for my project.
I can work on this if someone can help me for starter