balenasolutions/browser #31: All browser window can't be seen

yildirimmurat, 2020-10-14:
Hey All,
I am using this block as a service on my Raspberry Pi 4 to get display from a custom url. However, I always get top left corner part of the website. Even when I try to change WINDOW_SIZE at start.sh bash file, that only makes the display bigger but, still get only a part of the site.
Is there any other configuration that I miss?

phil-d-wilson, 2020-10-21:
@yildirimmurat Hi - can you paste in your docker-compose? Your browser window doesn't look how I expect it from the block running on the Pi4 - is that a photo of the browser block running on the Pi?

batjeciftci, 2020-10-29:
@yildirimmurat any updates, I'm also facing this problem.

batjeciftci, 2020-11-02:
Fixed by setting window size variable to: 1920x1080

phil-d-wilson, 2020-12-16:
Closing this, as there's no reply from the OP. Feel free to contact me if you are stilling having this issue though.

closed 2020-12-16 by phil-d-wilson