balenasolutions/browser #33: Download chrome pop-up (?) after a few hours

anujdeshpande, 2020-10-23:
A user sees a download chrome pop up after a few hours. I see this is a known "bug"

- https://support.google.com/chrome/thread/11537837?hl=en

I am trying to reproduce this locally, but not sure how many hours it's going to take to get there. Also, I am wondering if this is a pop-up in the traditional sense of it, or is it some native google chrome thing. Because typically we can disable popups in the settings

gplusnoel, 2020-10-24:
Hi I have attached another log from this morning…
Yesterday I ran RPi for around 5 hours on 23rd with absolutely no problems.
This morning I started it up again(24th) at 8:59 and when I returned I was presented with a blank screen. The log goes until 10:57 and the dashboard says online only heartbeat not received…
Hope the log is of some help…
Thanks
[23.10.20 12:30:00 (+1000)  scheduler  crond[15]: U.log](https://github.com/balenablocks/browser/files/5432343/23.10.20.12.30.00.%2B1000.scheduler.crond.15.U.log)

phil-d-wilson, 2020-10-24:
@anujdeshpande @gplusnoel 

The logs attached show that you are running the balenaDash project - which until yesterday was not using the browser block (i.e. this repo). As such, this issue is not concerned with this block.

Also, I have been running the browser block for months on a PiDisplay, and so always on - and I have never seen the message appear.

I would suggest @gplusnoel that you update your application with the latest balenaDash code, and see if the issue happens again: https://github.com/balenalabs/balena-dash

closed 2020-10-24 by phil-d-wilson