balenasolutions/browser #37: Video playback tearing (RPI4)

smarthomeagentur, 2020-12-07:
Hi,
I testet a lot for Raspberry Pi 4 to get Video Playback with chrome. All the samples have the problem of tearing. What works is WebGL 2.0 so there is some kind of accelerated video support.
An Example for the tearing looks like this:
[image]

phil-d-wilson, 2020-12-16:
@smarthomeagentur do you have the GPU enabled or disabled? Does it make a difference if you change the setting?

smarthomeagentur, 2020-12-17:
its the same with enabled and disabled

phil-d-wilson, 2020-12-17:
Can you try increasing the GPU memory?

https://www.balena.io/docs/reference/OS/advanced/

smarthomeagentur, 2021-01-04:
HI @phil-d-wilson I did it with GPU Mem 256 and also the normal 128/64. I think all settings do the same for video FullHD Playback

klvnptr, 2021-03-19:
tearing happens to me as well. pretty much tried everything. increased memory, enabled gpu, etc. any idea?

phil-d-wilson, 2021-04-08:
@klvnptr @smarthomeagentur - I've just pushed a new version of this block - would you mind re-testing and letting me know if this is still an issue?

ColinTravis, 2025-04-08:
I've been having this issue as well on a Raspberry Pi 5 (2gb). Using the newest version of the image (as well as older versions). Doesn't matter if it's playing at 720 or 4k, video memory allocation also doesn't seem to matter. I've read a few blog posts here and there that suggest disabling the compositor in wayland can help, but I'm not certain how to go about doing that in this image.