balenasolutions/browser #43: V2: Pass EnVars into chromium session

phil-d-wilson, 2021-01-18:


piotr-cz, 2021-01-28:
As far as I know there are very limited options:

- Append variables to `LAUNCH_URL` query
- Use launcher like [Playwright](https://github.com/microsoft/playwright), [Puppeteer](https://github.com/puppeteer/puppeteer) or Electron and set global variables
- Develop custom [Chromium pre-installed Extension](https://www.chromium.org/administrators/pre-installed-extensions) that sets global variables

phil-d-wilson, 2021-01-28:
Hi @piotr-cz 

Thanks for the info. I'm about to push a PR which will sort this, though. I can pass the enVars from the main container into the startx.sh file by whitelisting them in the `su` command.

Thanks

piotr-cz, 2021-01-28:
Can I ask how are you going to pass them to Chromium?

phil-d-wilson, 2021-01-28:
The majority of them get turned into Chromium flags:

https://github.com/balenablocks/browser/blob/9e8b3b17bb358a19c65be9d44d36b940a3bed31c/src/server.js#L110

@piotr-cz

piotr-cz, 2021-01-28:
Ah, I totally missed development on the V2 branch and the fact that it uses `chrome-launcher`.

Actually I'm looking for a way to pass data (environment variables) to global window context in which LAUNCH_URL is executed.

closed 2021-04-06 by phil-d-wilson