balenasolutions/browser #45: V2: Make Chromium pre-installed extensions configurable

piotr-cz, 2021-01-28:
Now two Chromium extensions are preinstalled using the [policy.json](https://github.com/balenablocks/browser/blob/97e68c3ebb88c9c5d3c50df2da17775d4d9ee44a/policy.json) file:

- [h264ify](https://chrome.google.com/webstore/detail/h264ify/aleakchihdccplidncghkekgioiakgal)
- [Adblock Plus](https://chrome.google.com/webstore/detail/adblock-plus-free-ad-bloc/cfhdojbkjhnklbpkdaibdccddilifddb)

I'm having problem with the second one.
At this moment it's [indirectly blocking any communication with sentry.io API](https://github.com/easylist/easylist/issues/6963) - which I use for error reporting.
Using error reporting tool is very important for me as some of the issues are unique for Chromium running on low CPU devices.

While for my use case it would be enough to omit the Adblock Plus extension, I imagine that other developers could benefit from option to add other extensions to the list.

phil-d-wilson, 2021-01-28:
@piotr-cz hey - yep this was on the project board but not an issue. I'll swap the dev to this issue, so that you can track it's progress.

piotr-cz, 2021-04-08:
@phil-d-wilson 
What's the recommended way to customize pre-installed extensions then?

I haven't seen any mentions in docs, is it possible with custom `policy.json` file as described in [Extend image configuration](https://github.com/balenablocks/browser#extend-image-configuration) with local display example?

phil-d-wilson, 2021-04-08:
Hey @piotr-cz - this was closed in error. It should be possible to install extensions with the policy.json file, but it's not working so far. For now, we have decided to push the new version of the browser block, but keep this issue open to track any efforts towards this.