balenasolutions/browser #53: Chrome doesn't use the full screen width

mbertheau, 2021-03-30:
Similar to what's mentioned in the readme https://github.com/balenablocks/browser/tree/4676a0230886b68afd613992a2039ee141423dc6#partial-display-output I'm experiencing **chrome not using the whole screen**. I noticed the following in the log:

```
30.03.21 19:00:26 (+0800)  kiosk  Running balena base image entrypoint...
30.03.21 19:00:27 (+0800)  kiosk  setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
30.03.21 19:00:27 (+0800)  kiosk  Using default chromium flags
30.03.21 19:00:27 (+0800)  kiosk  Enabling GPU acceleration
30.03.21 19:00:27 (+0800)  kiosk  Adding user settings directory
30.03.21 19:00:27 (+0800)  kiosk  Custom window size set to 
30.03.21 19:00:27 (+0800)  kiosk  Enabling kiosk mode
30.03.21 19:00:27 (+0800)  kiosk  Enabling cursor
```

Note the "Custom window size set to". This indicates that in the following code

https://github.com/balenablocks/browser/blob/4676a0230886b68afd613992a2039ee141423dc6/start.sh#L86-L93

the `if` doesn't take the right branch, even though I haven't configured a device service variable WINDOW_SIZE.

I'm hesitant to conclude that the particular `if` syntax doesn't do what it's supposed to do, given that it's used in that form in so many other places in https://github.com/balenablocks/browser/blob/4676a0230886b68afd613992a2039ee141423dc6/start.sh .

I observed that there's a device variable by the name `WINDOW_SIZE` in the balena dashboard. Maybe that interferes?

phil-d-wilson, 2021-04-06:
Hi there @mbertheau - we finally merged version 2 of the block today. I've tested the window size configuration, so hopefully moving to this version will solve your problem. :)

phil-d-wilson, 2021-04-08:
I'll close this issue for now - but feel free to re-open if this is still problematic.

closed 2021-04-08 by phil-d-wilson