balenasolutions/browser #59: Rotate touch input with display

daniel-quantica, 2021-04-08:
When `ROTATE_DISPLAY` is set the output is rotated but input events from a touchscreen are not. 
The touchscreen I'm using is the 11.6inch Waveshare FHD screen with touch attached via USB.

phil-d-wilson, 2021-04-08:
Hi @daniel-quantica - can you let me know what device you are using (e.g. raspberry pi 4)?

Also - I'm intrigued by what you're building, if you don't mind sharing. :)

daniel-quantica, 2021-04-08:
Right now I'm using a Raspberry Pi 4 Compute Module on the official IO board. I could test with other hardware as well, if that helps you (Upboard and NUC).

All I can say is that what we're building will end up in a physical device but with a different touch display (confirmed as being balena compatible).

phil-d-wilson, 2021-04-08:
@daniel-quantica thanks - I'll try and replicate with a Pi4 and NUC - and let you know. I don't have your specific screen to test with - but that shouldn't be an issue.

ngmartins, 2021-08-04:
Hello, I am experiencing the same problem and despite I haven't found a solution yet,  I think I have some useful information. I am a noob myself, but maybe someone with a better knowledge of those topics can point us in the right direction 🤞.

my setup:

- I am using an Intel NUC, and portrait LG TV, and a touch film.
- I am running the browser block with the `ROTATE_DISPLAY` set to `right`.

The touch film works but the axis was wrong.

I decided to install a full GUI OS (also X11 based) to do some more testing - I followed [this tututial](https://www.balena.io/blog/running-a-desktop-manager-with-balena/).

With the OS configured in landscape mode (default) everything worked as expected, however when I configure the display settings to portrait mode the behavior was the same as the experienced on my setup using Browser Block.

Looking around for a while I was able to fix this:
1. Install `xinput`: `apt-get install xinput`
2. List the inputs: `xinput list`
3. Collect the touch input ID
4. Invert axis: `xinput set-prop <ID> “Evdev Axis Inversion” 0, 1`
5. Swap axes:  `xinput set-prop <ID> “Evdev Axes Swap” 1`

After this, the touch was working as respected.

⚠️ Keep in mind the operations 4 and 5 are the ones that allow me to accomplish the `right` rotation and they are different for other rotations.

Then I back to my Browser Block setup. I SSH into the Browser container and installed `xinput`. However, I cannot run `xinput` commands. I am getting: `Unable to connect to X server`.

Then I back to the GUI OS setup and try to run the same command using SSH and I get the same result.
I think I should use the `-X` flag in the SSH command but I don't know how to use the flag with Balena.

Any thoughts?