balenasolutions/browser #64: Latest image from Docker registry contains v1 code

cailloumajor, 2021-04-25:
I try to use this balenablock as described in `readme.md` (docker-compose file), but on the Balena device (Raspberry Pi 4), the image I get from Docker registry ([balenablocks/browser:latest](https://hub.docker.com/layers/balenablocks/browser/latest/images/sha256-bc2701ab9b55bf37d4e69516ea7cf4f9504c0acff51295fd2be598929ea11fdf?context=explore), same as [balenablocks/browser:raspberrypi4-64](https://hub.docker.com/layers/balenablocks/browser/raspberrypi4-64/images/sha256-bc2701ab9b55bf37d4e69516ea7cf4f9504c0acff51295fd2be598929ea11fdf?context=explore)) does not contain current v2.0.1 code (https://github.com/balenablocks/browser/tree/v2.0.1), but v1.0.1 code (https://github.com/balenablocks/browser/tree/v1.0.1).

To finish with an idea, perhaps automated pushing of docker images from continuous delivery would allow easier review from the community and prevent such errors. I don't know if it is possible in Balena ecosystem.

phil-d-wilson, 2021-05-03:
Hey @cailloumajor - thanks for reaching out.

I have just pushed `balenablocks/browser` to a RaspberryPi4 and definitely got version 2 because it's exposing a Chromium remote debugging port.
We are just about to include the version number in the blocks output on startup, which will make things clearer.

To your idea: we are currently building a CI/CD pipeline internally, which will also build our blocks. I don't know the ETA for the work, but it is our highest priority stream of work currently.

closed 2021-05-04 by phil-d-wilson

oscarhult, 2021-05-12:
> I have just pushed `balenablocks/browser` to a RaspberryPi4 and definitely got version 2 because it's exposing a Chromium remote debugging port.

Please check `linux/arm64/v8` again, can't access the api (5011) or remote debugging (35173) on the latest pi4 image.