balenasolutions/browser #66: Please add support for Raspberry 2

itsbkm, 2021-04-30:
manifest for balenablocks/browser:raspberry-pi2 not found: manifest unknown: manifest unknown

phil-d-wilson, 2021-04-30:
@itsbkm - we could add this support - but honestly I don't know that the Chromium instances would work well enough to be useable on that processor and with that little memory. If you can get it to run, and it is usable do let us know.

itsbkm, 2021-04-30:
Yes it works fine,
We were using it in a kiosk mode at a restaurant to show the list of running orders.
chromium-browser --no-sandbox --no-first-run --noerrdialogs --start-fullscreen --start-maximized --disable-notifications --disable-infobars --kiosk --incognito 

Now we want to repurpose it to run Ads via slideshow

phil-d-wilson, 2021-05-04:
@itsbkm you're welcome to try. :)

I've pushed this image to my personal dockerhub org: https://hub.docker.com/layers/phildwilson/browser/raspberry-pi2/images/sha256-d26929fb1bb88384c0c252b237b93201f67d2ceaa11efd3124f4d06dbef5ac31?context=repo

You can add it to your compose file with the tag: `phildwilson/browser:raspberry-pi2`

If this works, let me know, and I can add it as a supported device on the balenaBlocks org. However, if it doesn't work, I don't currently have time to look into it - and would need community submissions to debug and resolve issues.

FirbyKirby, 2021-05-10:
I've been using the balenablocks browser image with a Raspberry Pi 2 and the [NetStatus](https://github.com/Ryandev/NetStatus) project successfully. The only thing that was necessary for the project to deploy well was to increase GPU memory from 16 MB to 128MB in the configuration variable (FBCP needed 64 MB to run, and balenablocks browser seemed to prefer 128 MB or more.) It's a little slow, but for this project, that's fine.

I'm happy to try out the version you posted @phil-d-wilson , though I don't know what it buys me over the latest standard balenablocks browser version. Can you enlighten me? Anything specifically optimized for Raspberry Pi 2? And assuming I should move forward with testing your image, can you merge PR #65 into it? As noted from issue #29 , I need this for proper local time display.

phil-d-wilson, 2021-05-11:
@FirbyKirby - No, I've not made any optimisations, just retagged the image for RPI2 and added it to the manifest, so that `latest` pulls the correct image for that device type.

Glad to here it works for you. 
`latest` will have all of the latest changes, including the TZ fix.