balenasolutions/browser #70: Chromium Remote Debugging

nslocum, 2021-05-20:
Has anyone gotten chromium remote debugging to work? I haven't had any luck.

I'm using an **rpi3** device
The port debug is open and exposed publicly on the device:
[image]
I can successfully connect to the other container on 8080.

Not sure what else I could be missing.

nslocum, 2021-05-20:
The `/flags` endpoint on the management API doesn't show the remote debugging flag:

    --disable-features=TranslateUI,--disable-component-extensions-with-background-pages,--disable-background-networking,--disable-sync,--metrics-recording-only,--disable-default-apps,--no-default-browser-check,--no-first-run,--disable-backgrounding-occluded-windows,--disable-renderer-backgrounding,--disable-background-timer-throttling,--force-fieldtrials=*BackgroundTracing/default/,--window-size=1920,1080,--window-position=0,0,--autoplay-policy=no-user-gesture-required,--noerrdialogs,--disable-session-crashed-bubble,--check-for-update-interval=31536000,--disable-dev-shm-usage,--disable-gpu

I added it manually using the `FLAGS` env variable, but it's still not working.

phil-d-wilson, 2021-05-20:
@nslocum I have used it during testing, so I know it did once work. I'll run it up again tomorrow and let you know if and how I did it.

Thanks for reaching out, it prompts me to document how to use it. 😊

nslocum, 2021-05-21:
@phil-d-wilson Thanks, any help you can provide would be greatly appreciated!

I can confirm that the remote debugging port is working. I can ssh to the local device and successfully run:

```sh
curl http://localhost:9222/json/version
curl http://localhost:9222
```

If the service is configured with `network_mode: host` in `docker-compose.yml`, then the debug port can be accessed from `host`. Otherwise, the debug port is only accessible from within the browser container.

nslocum, 2021-05-21:
@phil-d-wilson I got it working with your instructions fro the forum: https://forums.balena.io/t/balenadash-keeps-crashing-not-show-all-content/189902/16
the key part is to add the flags:
`--remote-debugging-address=0.0.0.0 --headless --remote-debugging-port=9222 --disable-dev-shm-usage --no-sandbox`

The only remaining issue is that the browser doesn't open the LAUNCH_URL when the flags are added.

I'm also trying to figure out a way to get it working without running Chrome in headless mode. It looks like it may be possible using port forwarding: https://embracethered.com/blog/posts/2020/chrome-spy-remote-control/

ebradbury, 2023-02-12:
I solved this by setting up an SSH tunnel on my local machine: `ssh -p 22222 -L 35173:localhost:35173 <balena user>@<device ip>`. You can get your username with `balena whoami`.

Then add `localhost:35173` to `chrome://inspect/#devices` and you should be able to inspect/control the browser tabs.