balenasolutions/browser #75: ROTATE_DISPLAY does not work on RPi 4

nmaas87, 2021-08-12:
The current version of the browser block does not rotate the display on the RPi 4 with the current settings / vc4-fkms-v3d.
If you decide to rotate, it will just resize the window - starting on 0,0 and then not e.g. landscape it like normal, but resize it in portrait if you ROTATE_DISPLAY=right - but not rotate the display/screen.
The only reliable way i found was to leave the ROTATE_DISPLAY and WINDOW_SIZE settings all alone and rotate the screen itself on boot by using BALENA_HOST_CONFIG_display_hdmi_rotate=3 (for 270 degree clockwise rotation). Then it will work flawlessly.

rahul-thakoor, 2021-09-29:
i think it might be related to https://github.com/balenablocks/browser/blob/a42c8c0da333a707b3a5e4b7b37e67a45e10a106/src/start.sh#L49

rahul-thakoor, 2021-11-03:
maybe fixed by #91 ?

Chrissi2812, 2022-01-06:
I have tested it with  **2.0.7** and it does work now.