balenasolutions/browser #77: Browser block does not run on rpi4

abhatikar, 2021-09-04:
Just a basic push and it does not come up

`[Logs]    [9/4/2021, 8:38:29 AM] Service exited 'browser sha256:fa7ca3bc2ac4314d5f68595bb8cb219c1b9259cffa39ae6f4617f1024543104c'
[Logs]    [9/4/2021, 8:39:21 AM] Restarting service 'browser sha256:fa7ca3bc2ac4314d5f68595bb8cb219c1b9259cffa39ae6f4617f1024543104c'
[Logs]    [9/4/2021, 8:39:22 AM] [browser] Running balena base image entrypoint...
[Logs]    [9/4/2021, 8:39:22 AM] [browser] setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
[Logs]    [9/4/2021, 8:39:27 AM] [browser] Using default chromium flags
[Logs]    [9/4/2021, 8:39:27 AM] [browser] Enabling GPU acceleration
[Logs]    [9/4/2021, 8:39:27 AM] [browser] Custom window size set to 800,600
[Logs]    [9/4/2021, 8:39:27 AM] [browser] Using default window position
[Logs]    [9/4/2021, 8:39:27 AM] [browser] Enabling kiosk mode
[Logs]    [9/4/2021, 8:39:27 AM] [browser] Enabling cursor
[Logs]    [9/4/2021, 8:39:31 AM] Service exited 'browser sha256:fa7ca3bc2ac4314d5f68595bb8cb219c1b9259cffa39ae6f4617f1024543104c'
[Logs]    [9/4/2021, 8:40:32 AM] Restarting service 'browser sha256:fa7ca3bc2ac4314d5f68595bb8cb219c1b9259cffa39ae6f4617f1024543104c'
[Logs]    [9/4/2021, 8:40:32 AM] [browser] Running balena base image entrypoint...
[Logs]    [9/4/2021, 8:40:33 AM] [browser] setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
[Logs]    [9/4/2021, 8:40:34 AM] [browser] Using default chromium flags
[Logs]    [9/4/2021, 8:40:34 AM] [browser] Enabling GPU acceleration
[Logs]    [9/4/2021, 8:40:34 AM] [browser] Custom window size set to 800,600
[Logs]    [9/4/2021, 8:40:34 AM] [browser] Using default window position
[Logs]    [9/4/2021, 8:40:34 AM] [browser] Enabling kiosk mode
[Logs]    [9/4/2021, 8:40:34 AM] [browser] Enabling cursor
[Logs]    [9/4/2021, 8:40:38 AM] Service exited 'browser sha256:fa7ca3bc2ac4314d5f68595bb8cb219c1b9259cffa39ae6f4617f1024543104c'`

pymoreau, 2021-09-04:
I have the same problem but it works if you downgrade the balenaOS to the v2.80.5+rev1

danthegoodman1, 2021-09-10:
@abhatikar Would you be able to share your docker compose file?

abhatikar, 2021-09-10:
As mentioned its the default docker-compose from the repo. However, reverting to previous release of balena os, the browser block works.

closed 2021-09-17 by bulldozer-balena[bot]

drewcovi, 2021-09-29:
@rahul-thakoor running `balenaOS 2.83.10+rev1` still seems to have this issue restarting... any specific image I should target in my docker-compose?

drewcovi, 2021-09-29:
```yml
  kiosk:
    image: balenablocks/browser:raspberrypi4-64
    privileged: true # required for UDEV to find plugged in peripherals such as a USB mouse
    network_mode: host
    ports:
        - '5011' # management API (optional)
        - '35173' # Chromium debugging port (optional)
    volumes:
      - 'settings:/data' # Only required if using PERSISTENT flag (see below)
```

```bash
 kiosk  Running balena base image entrypoint...
 kiosk  setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
 kiosk  Using default chromium flags
 kiosk  Disabling GPU acceleration
 kiosk  Adding user settings directory
 kiosk  Using fullscreen: 1824,984
 kiosk  Enabling kiosk mode
 kiosk  Disabling cursor
```

PR #83 (2021-09-17): Fix chromium not starting on balenaOS  v2.83.6+
[balenaOS v2.83.6](https://github.com/balena-os/meta-balena/blob/master/CHANGELOG.md#v2836) enables user namespacing by default, making applications such as chromium unable to run.
We override the kernel parameter `user.max_user_namespaces` in the application as a workaround.

Fixes #77 

Change-type: patch