balenasolutions/browser #82: Browser hangs unexpectedly

razmikarm, 2021-09-16:
We're working on a Jetson Nano with TFT display and are using browser container to show some stuff. We also use its API to configure what exactly needed to be shown. And from time to time it happens, that browser hangs and shows no changes on display, but on API level it shows, that everything works properly and browser is on a page you set with /url

phil-d-wilson, 2021-09-16:
Hey @razmikarm - do you get any error logs in balenaCloud dashboard when the crash happens?

razmikarm, 2021-09-16:
@phil-d-wilson Thank you for reply. No, seems like it even doesn't realize that.

phil-d-wilson, 2021-09-16:
@razmikarm if you remove TFT and use HDMI, can you recreate?

razmikarm, 2021-09-16:
It is actually connected from HDMI port.

phil-d-wilson, 2021-09-16:
@razmikarm To be clear, the TFT is powered from the nano, but connected to the HDMI port?

If so, can you then tell me the steps to recreate the crash, and I'll see if I can get my nano to do it.

razmikarm, 2021-09-16:
Actually I haven't found a reproduce way, it just happens unexpectedly, now We are working on it to find a way to reproduce this somehow.

phil-d-wilson, 2021-09-16:
@razmikarm excellent, thanks! If you find a way, let me know. I'll run mine up and see if it does it on it's own, but some repro steps are more likely to help me debug it.

razmikarm, 2021-09-16:
@phil-d-wilson, turns out problem is related to multiple tabs. We were using it in `KIOSK` mode and after exiting it we saw that there have been opened multiple tabs, which was unexpected for us, as we were only updating url sending `POST` request to `/url` endpoint (also would like to mention, that it is mentioned `PUT` method in documentation, but that method isn't allowed in server.js). Our system is fully automated, so when we tried to reproduce the issue manually, it turned out, that browser opens tabs without changing one tabs url, when make multiple requests in a row. Also it shows 'Restore tabs' chrome pop-up and refuses to response to any API call even after closing that pop-up window.