balenasolutions/browser #84: Docker version tags

edorgeville, 2021-09-22:
Hi,
We are trying to pin down our Docker image versions accross our projects. It would be greatly appreciated if this project could publish proper version tags. Currently, only `latest` and some platform specific tags are publish. We would like `2.0.4`, `2.0.5` to be available, for example.
Thank you!

phil-d-wilson, 2022-02-10:
Hi there @edorgeville 

I just wanted to let you know about some progress here.

In a nutshell, we want to be able to support version pinning with our blocks, but we've not wanted to put effort into scripts to build and maintain images on dockerhub. Balena Cloud has it's own container registry, since that's where your containers are running from when you do a `balena push` . So we want to be able to allow users to select which of their services are public, and then other users/fleets can use a human-readable slug to reference it in their dockerfile/compose. Blocks are also going to be made into a object type alongside fleets, so each time we push a release to balena cloud, users like you can reference a specific release.

We don't have deadlines or give estimates at balena, but I can tell you that one of our officially clever dudes (@klutchell ) is working on this as a priority, and it's beginning to feel like it's close now!

phil-d-wilson, 2022-04-21:
Hi @edorgeville - good news! You can now pin your block images by using the balena container registry:

https://www.balena.io/blog/improving-device-image-management-workflow-balenaHub-Container-Registry/

Let me know if you need any more help. :)

closed 2022-04-21 by phil-d-wilson