balenasolutions/browser #96: Example doesn't work anymore

titouanc, 2022-01-21:
Hello,

Since yesterday, this container `balenablocks/browser` doesn't seem to properly start anymore, even though I have seen it working 2 days ago

# Steps to reproduce

1. Create a new fleet in Balena (x86-64 qemu)
2. Download and launch a qemu image for the newly created fleet
3. Use the following `docker-compose.yml` (copied from the README of this repo): 
```yaml
version: '2'

volumes:
  settings:                          # Only required if using PERSISTENT flag (see below)

services:

  browser:
    image: balenablocks/browser
    privileged: true # required for UDEV to find plugged in peripherals such as a USB mouse
    network_mode: host
    ports:
        - '5011' # management API (optional)
        - '35173' # Chromium debugging port (optional)
    volumes:
      - 'settings:/data' # Only required if using PERSISTENT flag (see below)
```
4. `balena push mynewfleet`
5. The browser does not appear on screen, and in the logs from the Balena device page, I see:
```
21.01.22 10:23:08 (+0100) Restarting service 'browser sha256:8d26790c69ea8bd7363a8bcdc3b5fff1ea7f326dbc120dce58e8102000c6855f'
21.01.22 10:23:08 (+0100)  browser  user.max_user_namespaces = 10000
21.01.22 10:23:08 (+0100)  browser  umount: /dev: target is busy.
21.01.22 10:23:08 (+0100)  browser  mount: /sys/kernel/debug: nodev already mounted on /sys/kernel/debug.
21.01.22 10:23:08 (+0100)  browser  Running balena base image entrypoint...
21.01.22 10:23:08 (+0100)  browser  setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
21.01.22 10:23:08 (+0100)  browser  balenaBlocks browser version: 2.2.0
21.01.22 10:23:08 (+0100)  browser  Setting CPU Scaling Governor to 'performance'
21.01.22 10:23:08 (+0100)  browser  /usr/src/app/start.sh: line 19: /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor: No such file or directory
21.01.22 10:23:08 (+0100)  browser  Disabling cursor
21.01.22 10:23:08 (+0100)  browser  
21.01.22 10:23:08 (+0100)  browser  
21.01.22 10:23:08 (+0100)  browser  X.Org X Server 1.20.11
21.01.22 10:23:08 (+0100)  browser  X Protocol Version 11, Revision 0
21.01.22 10:23:08 (+0100)  browser  Build Operating System: linux Debian
21.01.22 10:23:08 (+0100)  browser  Current Operating System: Linux 2063378a1f6a 5.4.132-yocto-standard #1 SMP PREEMPT Wed Jul 14 20:01:00 UTC 2021 x86_64
21.01.22 10:23:08 (+0100)  browser  Kernel command line: BOOT_IMAGE=/boot/bzImage root=UUID=ba1eadef-a0bd-455b-89a6-b75055cfb11d rootwait console=ttyS0
21.01.22 10:23:08 (+0100)  browser  Build Date: 16 December 2021  05:08:23PM
21.01.22 10:23:08 (+0100)  browser  xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support) 
21.01.22 10:23:08 (+0100)  browser  Current version of pixman: 0.40.0
21.01.22 10:23:08 (+0100)  browser      Before reporting problems, check http://wiki.x.org
21.01.22 10:23:08 (+0100)  browser      to make sure that you have the latest version.
21.01.22 10:23:08 (+0100)  browser  Markers: (--) probed, (**) from config file, (==) default setting,
21.01.22 10:23:08 (+0100)  browser      (++) from command line, (!!) notice, (II) informational,
21.01.22 10:23:08 (+0100)  browser      (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
21.01.22 10:23:08 (+0100)  browser  (==) Log file: "/var/log/Xorg.0.log", Time: Fri Jan 21 09:23:08 2022
21.01.22 10:23:08 (+0100)  browser  (==) Using system config directory "/usr/share/X11/xorg.conf.d"
21.01.22 10:23:08 (+0100)  browser  pci id for fd 15: 1234:1111, driver (null)
21.01.22 10:23:08 (+0100)  browser  MESA-LOADER: failed to open bochs-drm: /usr/lib/dri/bochs-drm_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
21.01.22 10:23:08 (+0100)  browser  failed to load driver: bochs-drm
21.01.22 10:23:08 (+0100)  browser  (II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
21.01.22 10:23:08 (+0100)  browser  --- List Input Devices ---
21.01.22 10:23:08 (+0100)  browser  â¡ Virtual core pointer                     id=2    [master pointer  (3)]
21.01.22 10:23:08 (+0100)  browser  â   â³ Virtual core XTEST pointer                   id=4    [slave  pointer  (2)]
21.01.22 10:23:08 (+0100)  browser  â   â³ ImExPS/2 Generic Explorer Mouse              id=8    [slave  pointer  (2)]
21.01.22 10:23:08 (+0100)  browser  â£ Virtual core keyboard                    id=3    [master keyboard (2)]
21.01.22 10:23:08 (+0100)  browser      â³ Virtual core XTEST keyboard                  id=5    [slave  keyboard (3)]
21.01.22 10:23:08 (+0100)  browser      â³ Power Button                                 id=6    [slave  keyboard (3)]
21.01.22 10:23:08 (+0100)  browser      â³ AT Translated Set 2 keyboard                 id=7    [slave  keyboard (3)]
21.01.22 10:23:08 (+0100)  browser  ----- End of List --------
21.01.22 10:23:08 (+0100)  browser  Detecting window size from framebuffer
21.01.22 10:23:08 (+0100)  browser  Window size detected as 1024,768
21.01.22 10:23:08 (+0100)  browser  Browser API running on port: 5011
21.01.22 10:23:08 (+0100)  browser  Using LAUNCH_URL: http://app:5000
21.01.22 10:23:08 (+0100)  browser  Disabling GPU
21.01.22 10:23:08 (+0100)  browser  Enabling KIOSK mode
21.01.22 10:23:08 (+0100)  browser  Starting Chromium with flags: --disable-features=TranslateUI,--disable-component-extensions-with-background-pages,--disable-background-networking,--disable-sync,--metrics-recording-only,--disable-default-apps,--no-default-browser-check,--no-first-run,--disable-backgrounding-occluded-windows,--disable-renderer-backgrounding,--disable-background-timer-throttling,--force-fieldtrials=*BackgroundTracing/default/,--window-size=1024,768,--window-position=0,0,--autoplay-policy=no-user-gesture-required,--noerrdialogs,--disable-session-crashed-bubble,--check-for-update-interval=31536000,--disable-dev-shm-usage,--disable-gpu
21.01.22 10:23:08 (+0100)  browser  Displaying URL: --app= http://app:5000
21.01.22 10:23:33 (+0100)  browser  Main error:  Error: connect ECONNREFUSED 127.0.0.1:35173
21.01.22 10:23:33 (+0100)  browser      at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1144:16) {
21.01.22 10:23:33 (+0100)  browser    errno: 'ECONNREFUSED',
21.01.22 10:23:33 (+0100)  browser    code: 'ECONNREFUSED',
21.01.22 10:23:33 (+0100)  browser    syscall: 'connect',
21.01.22 10:23:33 (+0100)  browser    address: '127.0.0.1',
21.01.22 10:23:33 (+0100)  browser    port: 35173
21.01.22 10:23:33 (+0100)  browser  }
21.01.22 10:23:33 (+0100)  browser  xinit: connection to X server lost
21.01.22 10:23:33 (+0100)  browser  
21.01.22 10:23:33 (+0100)  browser  waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
21.01.22 10:23:35 (+0100) Service exited 'browser sha256:8d26790c69ea8bd7363a8bcdc3b5fff1ea7f326dbc120dce58e8102000c6855f'
```

phil-d-wilson, 2022-01-21:
Hey @titouanc 

Thanks for the repro - I'm pushing the block to a device now, and I'll take a look when I can.

It appears to be failing to connect to the debug port, which is odd. I suspect it will be something changed in Chromium.

phil-d-wilson, 2022-01-21:
Strangely @titouanc I cannot recreate the problem at all - the block is working fine for me on a PiZero2, Pi3 and Pi4.

titouanc, 2022-01-21:
@phil-d-wilson I just tested on a Raspberry Pi 3 (64b) and it does indeed work correctly. Could it be a problem with the x86-64 image only ?

When looking at `https://registry.hub.docker.com/v2/repositories/balenablocks/browser/tags`, I see that the x86-64 image has been updated yesterday, which matches the moment I started having problems.

phil-d-wilson, 2022-01-21:
@titouanc ah, yes, it could be! I'll run it up on my NUC and see what I find.

At the moment we don't have versioned images for our blocks - but we are addressing that.

Thanks for the patience. ;)

phil-d-wilson, 2022-01-24:
@titouanc - sorry for the delay. I have managed to recreate it on a NUC, so I'll try and investigate what's happening here.

Phil

phil-d-wilson, 2022-01-24:
@titouanc I've been hacking on this for hours, but I can't find anything more concrete than the issue being caused by the bump to Chromium v95. I'm testing pinning the version of Chromium to V90 which is working on all device types in testing.

phil-d-wilson, 2022-01-24:
@titouanc I'm just pushing version 2.3.0 to dockerhub. Would you mind testing your end before I close this issue? I've tested it on Pi3, Pi4 and a NUC and it's working for me. I've also echo'ed the Chromium version to the logging.

This should really cut down on the potential issues caused when we push a new version of the block - however, we are implementing pinned versions of the blocks. The reason we are not doing this now on Dockerhub, is because we want to move away from using Dockerhub for the blocks and use the balena container registry instead. This is a WIP but it is a focus of our work right now.

titouanc, 2022-01-24:
@phil-d-wilson I'll gladly do this as soon as it gets published to Dockerhub

phil-d-wilson, 2022-01-24:
@titouanc hey mate - the version 2.3.1 is up on dockerhub. I'm grateful for the help with testing. :)

titouanc, 2022-01-24:
I do not encounter the issue anymore, thank you for taking care of this @phil-d-wilson

phil-d-wilson, 2022-01-25:
Thanks for the help @titouanc

closed 2022-01-25 by phil-d-wilson