balenasolutions/browser #102: Ability to enable screensaver/display power down?

Zebble, 2022-01-30:
We're testing this using KIOSK mode and it works brilliantly.  Is there a way to have the screen powered down after a certain amount of time?

phil-d-wilson, 2022-01-30:
Hey @Zebble - not with this block, but I was going to chat to @rahul-thakoor about making the scheduler service from balena dash into another block, so that projects could use it. I have numerous devices around the house running this browser block, but currently they are on 24/7.

Zebble, 2022-01-30:
Thanks @phil-d-wilson - I did some digging into the scheduler service and that does seem like it will work by time, but possibly not by activity? (ie. 5 minutes after keyboard/mouse/touchscreen, turn off the screen).

I was able to modify the entrypoint/command through docker-compose for the browser service to do what I think will work for now.

    entrypoint: bash
    command: -c "(/usr/src/app/start.sh &) && sleep 15 && export XAUTHORITY=/home/chromium/.Xauthority && export DISPLAY=:0 && xset s 600 600 && xset +dpms && while true; do sleep 60; done;"

phil-d-wilson, 2022-01-30:
Oh NICE!

Zebble, 2022-02-03:
I made a slight change to my original workaround as this didn't seem to work when rolled out on balenaOS.  This command works:

command: -c "(/usr/src/app/start.sh &) && sleep 15 && XAUTHORITY=/home/chromium/.Xauthority DISPLAY=:0 && xset s 600 600 && XAUTHORITY=/home/chromium/.Xauthority DISPLAY=:0 xset +dpms && while true; do sleep 60; done;"

You will want to adjust the "600"'s to whatever screen timeout you need (in seconds).

The only remaining issue is that at least on an LCD touchscreen, touching the screen not only wakes up the screen but acts as an active press on whatever area you happen to touch.

Zebble, 2022-09-12:
Added a new XSET_COMMAND option in [PR #104](https://github.com/balenablocks/browser/pull/104) that has worked well for many months.  Hope this helps!