balenasolutions/browser #107: Not working on rpi4 with balenaOS 2.88.5+rev1

vbersier, 2022-02-22:
Hey there,

Trying to run a browser on a rpi4 device running balenaOS (latest as of writing) 64-bit, freshly flashed onto the SD.

Here's the docker-compose:

```yaml
version: '2.1'
services:
  browser:
    image: balenablocks/browser
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      - KIOSK=1
    depends_on:
      - some-other-service
```

And here is the output below. It seems the `vcgencmd` tool is not found and it can't determine the GPU memory (which is set to 512MB). This happens regardless if I specify the tag of the image as being `raspberrypi4-64` or not.

```
Running balena base image entrypoint...
setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
balenaBlocks browser version: 2.3.1
Setting CPU Scaling Governor to 'performance'
Disabling cursor
Checking GPU memory
VCHI initialization failed
/usr/src/app/start.sh: line 41: [: : integer expression expected
chown: cannot access '/data': No such file or directory
/usr/bin/startx: 69: cannot create /dev/null: Permission denied
/usr/bin/startx: 108: cannot create /dev/null: Permission denied
/usr/bin/startx: 132: cannot create /dev/null: Permission denied
/usr/bin/startx: 156: cannot create /dev/null: Permission denied
/usr/bin/startx: 200: cannot create /dev/null: Permission denied
xauth:  file /home/chromium/.Xauthority does not exist
/usr/bin/startx: 200: cannot create /dev/null: Permission denied
xauth: (stdin):1:  bad display name "ecb3fa4:0" in "add" command

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Raspbian
Current Operating System: Linux ecb3fa4 5.10.78-v8 #1 SMP PREEMPT Mon Nov 8 18:19:44 UTC 2021 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:22:EA:9B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 rootwait  vt.global_cursor_default=0 console=null cgroup_enable=memory root=UUID=ba1eadef-24c5-4f4e-939d-83ceb9439a9b rootwait
Build Date: 04 January 2022  10:24:42AM
xorg-server 2:1.20.11-1+rpt1+deb11u1 (https://www.debian.org/support) 
Current version of pixman: 0.40.0
    Before reporting problems, check http://wiki.x.org
    to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
    (++) from command line, (!!) notice, (II) informational,
    (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Feb 22 08:19:07 2022
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (No such file or directory)
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
     at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
xauth: (argv):1:  bad display name "ecb3fa4:0" in "remove" command
/usr/bin/startx: 235: cannot create /dev/null: Permission denied
Idling...
```

vbersier, 2022-02-22:
I notice now that the start.sh script is probably run as the root user and not the chromium user. Could that be the reason for this and the X11 stuff to fail? Does the root user have the video and tty groups assigned?

phil-d-wilson, 2022-02-23:
Hi there @vbersier 

I've just run up a new fleet with the browser block, on a Pi4 running OS 2.88.5+rev1 and everything works as expected:

[image]

As you can see, I can also run the GPU memory command fine. I don't think this is an issue with the block itself. 
You could ask for some more support on our forums, though, to see why your device isn't working as expected:

https://forums.balena.io/

closed 2022-02-23 by phil-d-wilson

vbersier, 2022-02-23:
Why does it say this when i got to add a new device to my fleet?

[image]

Seems like something is not right with that version. Also, could it be that we do not use the same revision of the rpi4 board? I have an older 1.2 revision board.

vbersier, 2022-02-23:
Confirmed working with a minimal setup and a new fleet. Will need to investigate