balenasolutions/browser #108: Browser cannot resolve mDNS hostnames

wjlove, 2022-03-11:
While the container configuration includes the network_mode: host option that appears to simply allow the container to access to the container host's network directly and does not effect how host names are resolved.  Since systemd.resolvd is not running in side this container DNS name resolution methods appear to be handled by /etc/nsswitch.conf and which currently suggests that only files and dns ("hosts:   files dns") be used to resolve host names and IPs.   In order for the browser block to resolve via  mDNS it would either need to run systemd.resolvd, which understands mDNS or include mdns4_minimal/mdns in the /etc/nsswitch.conf file.

dpslwk, 2025-05-01:
I'd like to use this block and have it access something like http://test-webserver.local

but I can not intuit form the above command exactly where and how I might "include mdns4_minimal/mdns in the /etc/nsswitch.conf file."

any help much appreciated, and maybe this can be then PR'd back

dpslwk, 2025-05-02:
ok looks like doing this with out another container would need this  feature adding to allow easy exposing of the host os Avahi socket `/var/run/avahi-daemon/socket`
https://github.com/balena-os/balena-supervisor/issues/1593
https://forums.balena.io/t/expose-avahi-socket-to-containers/373284/3

Dockerfile.template needs the following two extra lines to 
1) install `libnss-mdns` 
2) activate it in `/etc/nsswitch.conf`

```
RUN install_packages libnss-mdns
RUN sed -i 's/^hosts:          files dns/hosts:          files mdns4_minimal [NOTFOUND=return] dns mdns4/g' /etc/nsswitch.conf
```

This won't be able to resolve a .local address with out access to `/var/run/avahi-daemon/socket`

Thanks to this forum post I was able to add a `avahi` container along side the `browser` one
https://forums.balena.io/t/using-avahi-from-within-multi-container-project/368161/4?u=lwk