balenasolutions/browser #121: Raspberry Pi 4 GPU Acceleration only partially enabled?

sashahilton00, 2022-04-09:
Hi,

I have been testing this block out in the hopes of using if for a kiosk where hardware accelerated graphics are required. After launching the block with the following configuration:

```yaml
browser:
    image: bhcr.io/balenablocks/browser-aarch64
    privileged: true
    network_mode: host
    ports:
      - '5011'
    volumes:
      - 'browser-settings:/data'

    environment:
      LAUNCH_URL: 'https://testufo.com/'
      KIOSK: 0
      SHOW_CURSOR: 1
      ENABLE_GPU: 1
```

The `chrome://gpu` page is reporting that only rasterization is hardware accelerated - canvas, compositing and video decode are all software only, and opengl/webgl are unavailable all together. Please find attached a pastebin link for the full log from `chrome://gpu`: [https://pastebin.com/XkZbdS75](https://pastebin.com/XkZbdS75)

Is there something basic that I am missing? @rahul-thakoor commented on the balena forums a while back about needing to use a Pi specific build of chromium, but looking through this repo it looks as though any pi specific stuff has been removed/replaced by the generic aarch64 image. Thread is [here](https://forums.balena.io/t/raspberry-pi-4-gpu-acceleration-for-chromium-browser/36959/22). When trying the older, unmaintained balena blocks image hosted on Docker ([balenablocks/browser:raspberrypi4-64](https://hub.docker.com/layers/browser/balenablocks/browser/raspberrypi4-64/images/sha256-902ff6cfb0f7d2ed888ecf0be0e85280d768d9ef29689ae6e54686d6414ff245?context=explore)) the hardware acceleration is fully functional. A pastebin of the `chrome://gpu` report when running that block is available here: [https://pastebin.com/1B5TsJtz](https://pastebin.com/1B5TsJtz), which shows hardware acceleration working as expected.

Any thoughts on where I might be going wrong would be much appreciated.

rahul-thakoor, 2022-04-11:
Hello

In the current version, GPU hardware acceleration is disabled by default. You will need to set up `ENABLE_GPU=1` as an environment variable for the browser service.

rahul-thakoor, 2022-04-11:
You can find more details [here](https://github.com/balenablocks/browser#environment-variables)

sashahilton00, 2022-04-11:
Hi Rahul,

I have set that environment variable - the config at the top of my comment is the one that I am using.

rahul-thakoor, 2022-04-11:
Hi

Yes, we were originally using the following flags when enabling GPU
```
--enable-features=WebRTC-H264WithOpenH264FFmpeg 
--ignore-gpu-blacklist 
--enable-gpu-rasterization 
--force-gpu-rasterization 
--gpu-sandbox-failures-fatal=no 
--enable-native-gpu-memory-buffers
```
see [here](https://github.com/balenablocks/browser/blob/4676a0230886b68afd613992a2039ee141423dc6/start.sh#L60)

Now, we set the following

```
 --enable-zero-copy
--num-raster-threads=4
--ignore-gpu-blacklist
--enable-gpu-rasterization
```
as set [here](https://github.com/balenablocks/browser/blob/fce4881909a056a183c421dc58f19e311ae4c01f/src/server.js#L135)

You can use the `FLAGS` var to try and find a working configuration for you. 

Please note that this will overwrite the default flags we are using.

Can you please provide more details about your setup?
- which balenaOS version are you using?
- do you have any dtoverlays or config variable set?
- can you ssh into the hostOS and provide the output of `cat /mnt/boot/config.txt`?
- can you provide some output logs when the block starts?

Thanks

sashahilton00, 2022-04-12:
I'll run some tests with the flags provided above. With regards to version and config, I'm running `balenaOS 2.95.8 (development)`, I have `vc4-kms-v3d` as a dtoverlay, and `"i2c_arm=on","spi=on","audio=on"` as dtparams. I'll provide some logs and the boot config when I run the tests.

Marcocanc, 2022-05-17:
Hey all, I'm having similar issues.
Interestingly I had no issues with WebGL on RPi3. On RPi4 I can't get it to work (trying to render https://get.webgl.org/).

Chrissi2812, 2022-07-13:
This is caused by the version change in b2dd1a2067c561003b55a4bb5cd88d09df115a43.

I have tested two versions of the browser block on a Raspberry Pi 4.

|version|Chrome Version|WebGL|
|:-:|---|:-:|
|2.2.0|Chromium 92.0.4515.98 Built on Raspbian , running on Debian 10|✅|
|2.4.3|Chromium 89.0.4389.114 built on Debian 10.9, running on Debian 10.6|❌|

Seems newer versions of the browser-block use a generic chromium build. And this causes issues.

<details>
  <summary>Host Config</summary>

```
BALENA_HOST_CONFIG_disable_overscan 1
BALENA_HOST_CONFIG_gpu_freq         700
BALENA_HOST_CONFIG_gpu_mem          256
BALENA_HOST_CONFIG_over_voltage     4
RESIN_HOST_CONFIG_arm_64bit         1
RESIN_HOST_CONFIG_avoid_warnings    1
RESIN_HOST_CONFIG_disable_splash    1
RESIN_HOST_CONFIG_dtoverlay         "vc4-fkms-v3d"                   
RESIN_HOST_CONFIG_dtparam           "i2c_arm=on","spi=on","audio=on"
```

</details>

hanyufoodles, 2022-11-28:
We have this issue with the `browser-aarch64` version `2.4.6`.

By changing the base image from `balenalib/%%BALENA_MACHINE_NAME%%-debian-node:${NODEJS_VERSION}-buster-run` to `balenalib/%%BALENA_MACHINE_NAME%%-node:${NODEJS_VERSION}-run`, I made the hardware acceleration working.

I seems the debian buster node base image doesn't come with the GPU acceleration support for RPI v4. I hope this can help.

phil-d-wilson, 2024-09-05:
If you use the `browser-arm7hf` version of the block, it uses the RaspberryPi specific version of chromium, which has patches to enable the GPU correctly.

closed 2024-09-05 by phil-d-wilson