balenasolutions/browser #124: Just a black screen on RPi4 with balenaOS > 2.80.5+rev1

gitroli, 2022-04-27:
I get no screen, just black, when using balenaOS > 2.80.5+rev1
I am using RPi4.
Is there any fix or idea for that?

CallumHopkins181, 2022-05-26:
We had a similar problem to you, try changing setting "vc4-fkms-v3d" instead of "vc4-kms-v3d" in your device's dt_overlay configuration.

gitroli, 2022-05-27:
I didn't help, same situation, screen, stays black.

CallumHopkins181, 2022-05-27:
Alright, what screen are you using? If you're using an external touchscreen then try booting the raspberry pi with a HDMI screen instead, this could help narrow the problems down a bit.

Also, are you able to view the terminal in the Balena dashboard? If you can then it would be helpful to see the logs.

Lastly, does the raspberry pi display the Balena logo and then go blank, or does it not output anything at all?

gitroli, 2022-05-27:
I tried Touch Monitor Hannspree and also normal TV with HDMI.

I can enable terminal, but I didnt try so far, because its running in kiosk mode.

On boot I see the balena logo on fullscreen.

CallumHopkins181, 2022-05-27:
Alright, I'll wait until you give me the logs, but the fact that you see the Balena logo rules out the possibility of an incorrect OS configuration.

gitroli, 2022-05-27:
which logs do you need exactly, can I copy the logs also from the balena webinterface, because on terminal with keyboard it is hard to copy

CallumHopkins181, 2022-05-27:
You can copy the logs from the Balena dashboard, they should be identical to the logs on the device.

gitroli, 2022-05-27:
`<kiosk> balenaBlocks browser version: 2.0.3

<kiosk> Setting CPU Scaling Governor to 'performance'

Started service 'photo-gallery sha256:a4ec037991c541fb13fe4511f7a5a8601d8ecd6cd979d4fdde5a6c935cb72eff'
Killing service 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
Service exited 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
Killed service 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
Installing service 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
Installed service 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
Starting service 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
Started service 'datadog-iot sha256:fe0bf38a9804c1c771be297ba6173fecddbbb0807921e3463fae7427427780b3'
<photo-gallery> 

<photo-gallery> > album-slideshow@1.0.0 start /usr/src/app

<photo-gallery> > node server.js

<photo-gallery> 

<kiosk> Disabling cursor

<kiosk> Checking GPU memory

<datadog-iot> 2022-05-27 19:27:57 UTC | CORE | INFO | (pkg/util/log/log.go:596 in func1) | runtime: final GOMAXPROCS value is: 4

<datadog-iot> 2022-05-27 19:27:57 UTC | CORE | INFO | (pkg/util/log/log.go:596 in func1) | Features detected from environment: docker

<datadog-iot> 2022-05-27 19:27:57 UTC | CORE | INFO | (cmd/agent/app/run.go:245 in StartAgent) | Starting Datadog Agent v7.31.1

<kiosk> xauth:  file /home/chromium/.Xauthority does not exist

<kiosk> 

<kiosk> 

<kiosk> X.Org X Server 1.20.4

<kiosk> X Protocol Version 11, Revision 0

<kiosk> Build Operating System: Linux 5.10.17-v8+ aarch64 Raspbian

<kiosk> Current Operating System: Linux 86377dbbe4d6 5.10.95-v8 #1 SMP PREEMPT Thu Feb 17 11:43:01 UTC 2022 aarch64

<kiosk> Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=E4:5F:01:1A:C0:1E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 rootfstype=ext4 rootwait dwc_otg.lpm_enable=0 rootwait vt.global_cursor_default=0 console=null cgroup_enable=memory root=UUID=ba1eadef-963b-485f-8238-b87ede216464 rootwait

<kiosk> Build Date: 19 April 2021  03:23:36PM

<kiosk> xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support) 

<kiosk> Current version of pixman: 0.36.0

<kiosk> 	Before reporting problems, check http://wiki.x.org

<kiosk> 	to make sure that you have the latest version.

<kiosk> Markers: (--) probed, (**) from config file, (==) default setting,

<kiosk> 	(++) from command line, (!!) notice, (II) informational,

<kiosk> 	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.

<kiosk> (==) Log file: "/var/log/Xorg.0.log", Time: Fri May 27 19:27:57 2022

<kiosk> (==) Using system config directory "/usr/share/X11/xorg.conf.d"

<datadog-iot> 2022-05-27 19:27:58 UTC | CORE | INFO | (cmd/agent/app/run.go:314 in StartAgent) | Hostname is: 903c0c5

<kiosk> (II) modeset(0): Initializing kms color map for depth 24, 8 bpc.

[... 103 lines cut ...]

<datadog-iot> 2022-05-27 19:28:07 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:37 in CheckStarted) | check:network | Running check...

<datadog-iot> 2022-05-27 19:28:07 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:56 in CheckFinished) | check:network | Done running check

<datadog-iot> 2022-05-27 19:28:07 UTC | CORE | INFO | (pkg/serializer/serializer.go:371 in sendMetadata) | Sent metadata payload, size (raw/compressed): 4637/1860 bytes.

<datadog-iot> 2022-05-27 19:28:07 UTC | CORE | INFO | (pkg/forwarder/transaction/transaction.go:365 in internalProcess) | Successfully posted payload to "https://7-31-1-app.agent.datadoghq.eu/intake/?api_key=***************************78f5e", the agent will only log transaction success every 500 transactions

<datadog-iot> 2022-05-27 19:28:08 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:37 in CheckStarted) | check:load | Running check...

<datadog-iot> 2022-05-27 19:28:08 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:56 in CheckFinished) | check:load | Done running check

<datadog-iot> 2022-05-27 19:28:09 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:37 in CheckStarted) | check:disk | Running check...

<datadog-iot> 2022-05-27 19:28:09 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:56 in CheckFinished) | check:disk | Done running check

<datadog-iot> 2022-05-27 19:28:14 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:37 in CheckStarted) | check:uptime | Running check...

<datadog-iot> 2022-05-27 19:28:14 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:56 in CheckFinished) | check:uptime | Done running check

<datadog-iot> 2022-05-27 19:28:15 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:37 in CheckStarted) | check:memory | Running check...

<datadog-iot> 2022-05-27 19:28:15 UTC | CORE | INFO | (pkg/collector/worker/check_logger.go:56 in CheckFinished) | check:memory | Done running check
`

gitroli, 2022-05-27:
Here is only the kiosk (balenablock browser) filtered:

gitroli, 2022-05-27:
kiosk  Starting Chromium with flags: --disable-features=TranslateUI,--disable-component-extensions-with-background-pages,--disable-background-networking,--disable-sync,--metrics-recording-only,--disable-default-apps,--no-default-browser-check,--no-first-run,--disable-backgrounding-occluded-windows,--disable-renderer-backgrounding,--disable-background-timer-throttling,--force-fieldtrials=*BackgroundTracing/default/,--window-size=1920,1080,--window-position=0,0,--autoplay-policy=no-user-gesture-required,--noerrdialogs,--disable-session-crashed-bubble,--check-for-update-interval=31536000,--disable-dev-shm-usage,--enable-zero-copy,--num-raster-threads=4,--ignore-gpu-blacklist,--enable-gpu-rasterization,--user-data-dir=/data/chromium
 kiosk  Displaying URL: --app= http://google.at
 kiosk  (node:385) UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 127.0.0.1:35173
 kiosk      at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1144:16)
 kiosk  (node:385) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
 kiosk  (node:385) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

CallumHopkins181, 2022-05-27:
I'm not a JavaScript expert, but it appears as if the browser is running into an unhandled exception:
`UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 127.0.0.1:35173`

Have you tried setting the `LAUNCH_URL` variable to nothing, so that it will attempt to display the default html page?

gitroli, 2022-05-29:
It seems the issue is known:
https://github.com/balena-os/meta-balena/issues/2319

gitroli, 2022-05-29:
And here:
https://github.com/balenablocks/browser/issues/77

gitroli, 2022-05-29:
but I find no solution

CallumHopkins181, 2022-05-29:
Is there any particular reason you need to use such an outdated OS? The latest version is 2.98.11!

gitroli, 2022-05-29:
yes I am using 2.98.11, it is not working with this version, chromium is only working when balena OS <= 2.85

vbersier, 2022-06-22:
> We had a similar problem to you, try changing setting "vc4-fkms-v3d" instead of "vc4-kms-v3d" in your device's dt_overlay configuration.

This has helped me. I was having a black screen and changed the dt_overlay and now it works for me on the latest stable balenaOS.

beeb, 2022-06-23:
> I tried this already, but it didnt help.

I had issues when another container on my device was trying to use udev for USB access. Do you have other containers with UDEV=on?

gitroli, 2022-06-23:
No

gitroli, 2022-06-23:
I think this is the problem:
It seems the issue is known:
https://github.com/balena-os/meta-balena/issues/2319

closed 2024-09-04 by phil-d-wilson