balenasolutions/browser #135: Chromium-common 89.0.4389.114-1~deb10u1 not found in latest image

beda, 2022-09-28:
I get the following error when I try to build browser:
E: Version '89.0.4389.114-1~deb10u1' for 'chromium-common' was not found

When I update install_packages the build is working:
`RUN install_packages \
    chromium-common \
    chromium \`

I'm not sure if there is a requirement for version 89

Iznogohul, 2022-11-10:
Hello @beda, 

Version `89.0.4389.114-1~deb10u1` is targeted only for arm64 architecture so I assume you are trying to build it with a machine that has some other architecture.

Using this `RUN install_packages \ chromium-common \ chromium \` without specific which version you want to install will make the build work because it is trying to install the latest available package for your architecture and distro(buster in this case).
Using 90.0.4430.212-1~deb10u1 for some time now and didn't stumble across any major issues yet.

[Information about chromium package here](https://packages.debian.org/search?keywords=chromiuml)

nucleardreamer, 2022-11-16:
@Iznogohul this should now be resolved for this version! Thank you for the shout out, I believe that the unpinned packages from `buster` wont be moving for a while either. I opened an issue for us to move to bullseye here: #138 but it might take some time to test. CI is auto publishing this release now!

cc @beda

closed 2022-11-16 by nucleardreamer

beda, 2022-11-16:
I did some testing with an Intel NUC. I had some issues with videos stuttering in chromium with buster and chromium 90. 

I got better results by switching to bullseye and chromium 107. I'm not yet 100% sure what is required. I enabled gpu but there might be some additional flags that need to be set with chromium above 90.
`
          '--use-gl=desktop',
          '--enable-features=VaapiVideoDecoder'
`