balenasolutions/browser #141: Cannot pull image anymore due to platform mismatch.

jgeskens, 2023-03-01:
After months of working great, we suddenly get this error during our "balena push":

[Error]    Some services failed to build:
[Error]      Service: browser
[Error]        Error: failed to get destination image "sha256:153e5abf0f310c860ab725ab54441f2f6f92e8dd6a8a8c71a571c8b92526c3cf": image with reference sha256:153e5abf0f310c860ab725ab54441f2f6f92e8dd6a8a8c71a571c8b92526c3cf was found but does not match the specified platform: wanted linux/arm64/v8, actual: linux/amd64
[Info]     Built on arm02

We want to build for aarch64.

What the heck is going on here?

phil-d-wilson, 2023-03-01:
What the heck indeed!

Can you show me your dockerfile?

jgeskens, 2023-03-01:
```
FROM bh.cr/balenalabs/browser-%%BALENA_ARCH%%

# Install supervisord
RUN set -ex \
    && apt-get update \
    && apt-get install --no-install-recommends -y supervisor \
    && apt-get autoremove \
    && rm -rf /var/lib/apt/lists/*

# Add requirements
COPY requirements.txt /tmp/requirements.txt

# Install requirements
RUN set -ex \
    && BUILD_DEPS=" \
    python3-dev \
    python3-pip \
    build-essential \
    git-core" \
    && apt-get update \
    && apt-get install --no-install-recommends -y ${BUILD_DEPS} \
    && pip3 install -U pip wheel \
    && pip3 install -r /tmp/requirements.txt \
    && apt-get purge -y --auto-remove ${BUILD_DEPS} \
    && rm -rf /root/.cache/* \
    && rm -rf /var/lib/apt/lists/*

# Define environment
ENV INITSYSTEM=on

# Create mini ram cache directory
RUN echo "tmpfs  /ramcache  tmpfs  rw,size=10M,nosuid,nodev,noexec  0 0" >> /etc/fstab

# Set entrypoint
COPY docker-entrypoint.sh /usr/local/bin/entrypoint.sh
RUN chmod +x /usr/local/bin/entrypoint.sh
ENTRYPOINT ["/usr/local/bin/entrypoint.sh"]

# Copy the wait logic
COPY wait-for-host.sh /usr/src/app/wait-for-host.sh

# Copy supervisord and supervisor app configurations
COPY ./supervisor /etc/supervisor

# Copy scripts
COPY ./scripts /usr/src/scripts/

# Copy minion
COPY ./minion /usr/src/minion/

# Start supervisor
CMD ["/usr/bin/supervisord"]
```

I was pushing to a fleet with raspberrypicm4-ioboard as platform.

phil-d-wilson, 2023-03-01:
I'll try and replicate the issue tomorrow morning (UTC) and let you know what I find.

ztaylor4, 2023-03-01:
I am also experiencing this issue.  It has prevented me from being able to push new builds to balena cloud since I ran into it yesterday.

Here is a discussion which I believe is related:
https://forums.balena.io/t/balena-cli-suddenly-fail-the-requested-images-platform-linux-amd64-does-not-match-the-detected-host-platform/366897/31

And here is a very simple Dockerfile that will fail with this error:

```
FROM --platform=linux/arm64 balenalib/raspberrypi4-64-golang:1.19 as builder
RUN go install -v github.com/mozz100/tohora@v0.3.2
FROM  bh.cr/balenalabs/browser-aarch64
RUN install_packages cec-utils
COPY --from=builder /go/bin/tohora /home/chromium/tohora
```

phil-d-wilson, 2023-03-02:
Good news @ztaylor4 @jgeskens - I have found the issue and I'm pushing a fix.

In a nutshell, the block has been happily working for so long, it was using an old base image. This image was built in a time we had to hack around a limitation with the balena builder. That limitation was fixed properly, but old base images have the hack which now annoys the builder. I've simply updated the block to use a newer base image, and the issue is resolved (using both your repro dockerfiles - thank you for adding them, it saved me some minutes).

The browser block is now at version 2.4.7

To validate the fix you can repro the error by specifying the previous version:

`FROM bh.cr/balenablocks/browser-aarch64/2.4.6`

in your dockerfile, and then trying again with latest or specifying the new version:

`FROM bh.cr/balenablocks/browser-aarch64/2.4.7`

phil-d-wilson, 2023-03-02:
Fix pushed and validated

closed 2023-03-02 by phil-d-wilson