balenasolutions/browser #144: Touchscreen Strings

sstephanyAE, 2023-04-04:
What are the acceptable touchscreen strings?  Are they the same as rotate_display?

phil-d-wilson, 2023-04-04:
Hey, sorry I don't understand the question. What are touchscreen strings?

sstephanyAE, 2023-04-04:
The options for the environment variable TOUCHSCREEN.

[image]

phil-d-wilson, 2023-04-05:
Aha! Thanks for the context!

This was a community contributed feature (hence why it didn't ring bells in my head). The commit is here:

https://github.com/balena-labs-projects/browser/commit/f8404bf7496f952c7dc16b57cc59d4a98c8b6565

and the PR with some details is here:

https://github.com/balena-labs-projects/browser/pull/95

It looks like you either need to know what your touchscreen device name is, or find it using xinput.

Hope that helps.

sstephanyAE, 2023-04-05:
Thank you.  Unfortunately since I am using Balena everything is remote and xinput does not work.  I was able to do evtest, but it gave me a generic and not the actual name of the touchscreen.  Do you know of a way to get the touchscreen name while working remote?

Sarah

f8404bf<https://github.com/balena-labs-projects/browser/commit/f8404bf7496f952c7dc16b57cc59d4a98c8b6565>

#95<https://github.com/balena-labs-projects/browser/pull/95>

phil-d-wilson, 2023-04-06:
Hey Sarah,

The [startx.sh](https://github.com/balena-labs-projects/browser/blob/master/src/startx.sh) script runs xinput itself, so if you restart the container running the browser block, and watch the output, it will list your input devices. Here's my output from one of my devices:

```
06.04.23 08:00:05 (+0100)  browser  --- List Input Devices ---
06.04.23 08:00:05 (+0100)  browser  â¡ Virtual core pointer                     id=2    [master pointer  (3)]
06.04.23 08:00:05 (+0100)  browser  â   â³ Virtual core XTEST pointer                   id=4    [slave  pointer  (2)]
06.04.23 08:00:05 (+0100)  browser  â   â³ TSTP MTouch                                  id=6    [slave  pointer  (2)]
06.04.23 08:00:05 (+0100)  browser  â   â³ Microsoft MicrosoftÂ® 2.4GHz Transceiver v7.0 Mouse  id=8    [slave  pointer  (2)]
06.04.23 08:00:05 (+0100)  browser  â   â³ Microsoft MicrosoftÂ® 2.4GHz Transceiver v7.0 Consumer Control       id=9    [slave  pointer  (2)]
06.04.23 08:00:05 (+0100)  browser  â   â³ Microsoft MicrosoftÂ® 2.4GHz Transceiver v7.0 Consumer Control       id=10   [slave  pointer  (2)]
06.04.23 08:00:05 (+0100)  browser  â£ Virtual core keyboard                    id=3    [master keyboard (2)]
06.04.23 08:00:05 (+0100)  browser      â³ Virtual core XTEST keyboard                  id=5    [slave  keyboard (3)]
06.04.23 08:00:05 (+0100)  browser      â³ Microsoft MicrosoftÂ® 2.4GHz Transceiver v7.0        id=7    [slave  keyboard (3)]
06.04.23 08:00:05 (+0100)  browser      â³ Microsoft MicrosoftÂ® 2.4GHz Transceiver v7.0 System Control id=11   [slave  keyboard (3)]
06.04.23 08:00:05 (+0100)  browser  ----- End of List --------
```

does that give you what you need?

Phil

phil-d-wilson, 2023-04-06:
No problem. Let me know if you need any more help. 😊

closed 2023-04-06 by phil-d-wilson