balenasolutions/browser #146: Add feature to set headers in the request

oscni, 2023-05-12:
Would be very useful to be able to set headers in the request.

phil-d-wilson, 2024-09-04:
What headers in what request @oscni ?

oscni, 2024-09-07:
http headers in http requests

phil-d-wilson, 2024-09-08:
Sure. The block only makes requests itself to other services running on the device to automatically find what else it could display. Any actual requests are made through chrome-launcher which does not support request headers.

closed 2024-09-08 by phil-d-wilson

oscni, 2024-09-09:
yeah i noticed that it doesn't support it natively but I actually got it working by modifying this repo. It involved spinning up a small local API webserver which present the header information and then a chrome pluing that loads that header information at startup and in all requests uses that header information. Just thought it would be good if the feature is supported in balena-browser since it's useful in many setups.