balenasolutions/browser #149: Rotate Display not working for jetson

MaMiFreak, 2023-07-17:
We are trying to get a portrait mode screen working on a jetson but we are 
running into some troubles getting it to work.

All the provided options to rotate the screen do not result in it rotating and we are not sure on how to debug 
this any further. any suggestions on what might be going on?

system:
- NVIDIA Jetson Nano Developer Kit
  https://developer.nvidia.com/embedded/jetson-nano-developer-kit
- beetronics screen with hdmi
- balena version balenaOS 2.98.12
- balena browser-aarch64 
 
compose file
`
# Balena compose file support is currently based on version 2.4, as such any fielsd that were introduced in version 3 are not supoprted
# https://docs.balena.io/reference/supervisor/docker-compose/
version: "2.4"
services:
  browser:
    image: bh.cr/balenalabs/browser-aarch64 # where <arch> is one of aarch64, arm32 or amd64
    privileged: true # required for UDEV to find plugged in peripherals such as a USB mouse
    network_mode: host
    ports:
        - '5011' # management API (optional)
        - '35173' # Chromium debugging port (optional)

    environment: 
      LAUNCH_URL: '"http://google.com/"'
`

we also tested setting below without any change
ROTATE_DISPLAY	 left
BALENA_HOST_CONFIG_display_rotate 1

Full log of balena browser docker
`
user.max_user_namespaces = 10000
 browser  umount: /dev: target is busy.
 browser  Running balena base image entrypoint...
 browser  setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
 browser  balenaLabs browser version: 2.4.7
 browser  Setting CPU Scaling Governor to 'performance'
 browser  Enabling cursor
 browser  hostname: No address associated with hostname
 browser  xauth:  file /home/chromium/.Xauthority does not exist
 browser  xauth: (stdin):1:  bad display name "53c57d6:0" in "add" command
 browser  
 browser  
 browser  http://X.Org  X Server 1.20.4
 browser  X Protocol Version 11, Revision 0
 browser  Build Operating System: Linux 5.10.0-21-arm64 aarch64 Debian
 browser  Current Operating System: Linux 53c57d6 4.9.253-l4t-r32.7 #1 SMP PREEMPT Fri Mar 11 13:27:19 UTC 2022 aarch64
 browser  Kernel command line: tegraid=21.1.2.0.0 ddr_die=4096M@2048M section=512M memtype=0 vpr_resize usb_port_owner_info=0 lane_owner_info=0 emc_max_dvfs=0 touch_id=0@63 video=tegrafb no_console_suspend=1 console=ttyS0,115200n8 debug_uartport=lsport,4 earlyprintk=uart8250-32bit,0x70006000 maxcpus=4 usbcore.old_scheme_first=1 lp0_vec=0x1000@0xff780000 core_edp_mv=1075 core_edp_ma=4000 gpt tegra_fbmem=0x800000@0x92ca9000 is_hdmi_initialised=1  console=ttyS0,115200 console=tty0 fbcon=map:0 net.ifnames=0 root=UUID=ba1eadef-67c7-43db-890e-063b4f89f519 ro rootwait l4tver=32.7.1 console=null quiet splash vt.global_cursor_default=0 consoleblank=0 sdhci_tegra.en_boot_part_access=1
 browser  Build Date: 05 February 2023  09:03:02PM
 browser  xorg-server 2:1.20.4-1+deb10u8 (https://www.debian.org/support)
 browser  Current version of pixman: 0.36.0
 browser        Before reporting problems, check http://wiki.x.org 
 browser        to make sure that you have the latest version.
 browser  Markers: (--) probed, (**) from config file, (==) default setting,
 browser        (++) from command line, (!!) notice, (II) informational,
 browser        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
 browser  (==) Log file: "/var/log/Xorg.0.log", Time: Tue Jul  4 10:28:19 2023
 browser  (==) Using system config directory "/usr/share/X11/xorg.conf.d"
 browser  --- List Input Devices ---
 browser  â¡ Virtual core pointer                       id=2    [master pointer  (3)]
 browser  â   â³ Virtual core XTEST pointer                     id=4    [slave  pointer  (2)]
 browser  â   â³ Weida Hi-Tech CoolTouchÂ® System               id=6    [slave  pointer  (2)]
 browser  â   â³ SteelSeries SteelSeries Rival 310 eSports Mouse        id=7    [slave  pointer  (2)]
 browser  â   â³ SteelSeries SteelSeries Rival 310 eSports Mouse        id=8    [slave  pointer  (2)]
 browser  â£ Virtual core keyboard                      id=3    [master keyboard (2)]
 browser      â³ Virtual core XTEST keyboard                    id=5    [slave  keyboard (3)]
 browser      â³ gpio-keys                                      id=9    [slave  keyboard (3)]
 browser  ----- End of List --------
 browser  Window size set by environment variable to 800,1280
 browser  X Error of failed request:  BadMatch (invalid parameter attributes)
 browser    Major opcode of failed request:  140 (RANDR)
 browser    Minor opcode of failed request:  2 (RRSetScreenConfig)
 browser    Serial number of failed request:  14
 browser    Current serial number in output stream:  14
 browser  Display rotated to portait. Reversing screen coordinates
 browser  Reversed window size: 1280,800
 browser  Rotate Touch Inputs
 browser  Rotating Input (6): left
 browser  Installed browser version: Chromium 89.0.4389.114 built on Debian 10.9, running on Debian 10.12
 browser  Browser API running on port: 5011
 browser  Using LAUNCH_URL: http://google.com/ 
 browser  Disabling GPU
 browser  Enabling KIOSK mode
 browser  Starting Chromium with flags: --disable-features=TranslateUI,--disable-component-extensions-with-background-pages,--disable-background-networking,--disable-sync,--metrics-recording-only,--disable-default-apps,--no-default-browser-check,--no-first-run,--disable-backgrounding-occluded-windows,--disable-renderer-backgrounding,--disable-background-timer-throttling,--force-fieldtrials=*BackgroundTracing/default/,--window-size=1280,800,--window-position=100,0,--autoplay-policy=no-user-gesture-required,--noerrdialogs,--disable-session-crashed-bubble,--check-for-update-interval=31536000,--disable-dev-shm-usage,--disable-gpu
 browser  Displaying URL: --app= http://google.com/ 
 browser  Chromium remote debugging tools running on port: 35173
Killing service 'browser sha256:7b466dd4970d46dedb5d8b06595b718802a0e110920ca7d0c0ef1f67a8c9c9de'
Killing service 'browser sha256:7b466dd4970d46dedb5d8b06595b718802a0e110920ca7d0c0ef1f67a8c9c9de'
 browser  --- List Input Devices ---
 browser  â¡ Virtual core pointer                       id=2    [master pointer  (3)]
 browser  â   â³ Virtual core XTEST pointer                     id=4    [slave  pointer  (2)]
 browser  â   â³ Weida Hi-Tech CoolTouchÂ® System               id=6    [slave  pointer  (2)]
 browser  â   â³ SteelSeries SteelSeries Rival 310 eSports Mouse        id=7    [slave  pointer  (2)]
 browser  â   â³ SteelSeries SteelSeries Rival 310 eSports Mouse        id=8    [slave  pointer  (2)]
 browser  â£ Virtual core keyboard                      id=3    [master keyboard (2)]
 browser      â³ Virtual core XTEST keyboard                    id=5    [slave  keyboard (3)]
 browser      â³ gpio-keys                                      id=9    [slave  keyboard (3)]
 browser  ----- End of List --------
 browser  Window size set by environment variable to 1280,800
 browser  Chromium remote debugging tools running on port: 35173
 `

MiroYld, 2023-11-20:
Hello @MaMiFreak ,
Do you try to add Weida Hi-Tech CoolTouchÂ® System to TOUCHSCREEN  variable

phil-d-wilson, 2024-09-04:
@MaMiFreak do you still have this issue?

MaMiFreak, 2024-09-13:
> @MaMiFreak do you still have this issue?

A few months ago i tested this and the problem persisted, i'll try again with the newest version to verify this. 
we added some custom css rotations and limited the design in some places and managed to get it functional on our
target device

phil-d-wilson, 2024-09-13:
Thanks @MaMiFreak - let me know what you find. If you still have the problem, then I'll try to repro the issue. I have a Jetson Orin Nano which I can test with in the coming weeks.

MiroYld, 2024-09-13:
Hello @MaMiFreak,
I think this script can help you
```bash
#!/usr/bin/env bash

#Disable screen blanking
xset s off -dpms

echo "--- List Input Devices ---"
xinput list
echo "----- End of List --------"

function reverse_window_coordinates () {
  local INPUT=$1

  IFS=', ' read -a coords <<< $INPUT
  if [ ${#coords[@]} -eq 2 ]; then
    echo "${coords[1]},${coords[0]}"
  else
    echo "Screen coordinates not set correctly, so cannot reverse them."
  fi
}

function rotate_touch () {
  echo "Rotating Input ($1): $2"
  local TRANSFORM='Coordinate Transformation Matrix'

  case "$2" in
    normal)   xinput set-prop "$1" "$TRANSFORM" 1 0 0 0 1 0 0 0 1;;
    inverted) xinput set-prop "$1" "$TRANSFORM" -1 0 1 0 -1 1 0 0 1;;
    left)     xinput set-prop "$1" "$TRANSFORM" 0 -1 1 1 0 0 0 0 1;;
    right)    xinput set-prop "$1" "$TRANSFORM" 0 1 0 -1 0 1 0 0 1;;
  esac
}

# set whether to show a cursor or not
if [[ ! -z $SHOW_CURSOR ]] && [[ "$SHOW_CURSOR" -eq "1" ]]
  then
    export CURSOR=''
    echo "Enabling cursor"
  else
    export CURSOR='-- -nocursor'
    echo "Disabling cursor"
fi

if [[ -z "$WINDOW_SIZE" ]]; then
  # detect the window size from the framebuffer file
  export WINDOW_SIZE=$( cat /sys/class/graphics/fb0/virtual_size )
  echo "Window size detected as $WINDOW_SIZE"
else
  echo "Window size set by environment variable to $WINDOW_SIZE"
fi

# rotate screen if env variable is set [normal, inverted, left or right]
if [[ ! -z "$ROTATE_DISPLAY" ]]; then
  ROTATE_DELAY="${ROTATE_DELAY:-3}"

  sleep "$ROTATE_DELAY" && xrandr -o $ROTATE_DISPLAY

  #If the display is rotated to left or right, we need to reverse the size and position coords
  if [[ "$ROTATE_DISPLAY" == "left" ]] || [[ "$ROTATE_DISPLAY" == "right" ]]; then

    echo "Display rotated to portait. Reversing screen coordinates"

    #window size
    REVERSED_SIZE="$(reverse_window_coordinates $WINDOW_SIZE)"
    WINDOW_SIZE=$REVERSED_SIZE
    echo "Reversed window size: $WINDOW_SIZE"

    #window position, if set
    if [[ "$WINDOW_POSITION" -ne "0,0" ]]
    then
      REVERSED_POSITION="$(reverse_window_coordinates $WINDOW_POSITION)"
      export WINDOW_POSITION=$REVERSED_POSITION
      echo "Reversed window position: $WINDOW_POSITION"
    fi
  fi

  echo "Rotate Touch Inputs"
  if [[ ! -z "$TOUCHSCREEN" ]]; then
    rotate_touch "$TOUCHSCREEN" "$ROTATE_DISPLAY"
  else
    devices=$( xinput --list | fgrep Touch | sed -E 's/^.*id=([0-9]+).*$/\1/' )
    for device in ${devices} ;do
	rotate_touch $device $ROTATE_DISPLAY
    done
  fi
fi

setxkbmap -option "terminate:ctrl_alt_bksp"

matchbox-window-manager -use_titlebar no &
unclutter -idle 0.5 &    # hide X mouse cursor unless mouse activated

# Disable crash status in chromium
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/pi/.config/chromium/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"Normal"/' /home/pi/.config/chromium/Default/Preferences

URLS="https://github.com/MiroYld"
chromium-browser $URLS \
  --enable-features=OverlayScrollbar,OverlayScrollbarFlashAfterAnyScrollUpdate,OverlayScrollbarFlashWhenMouseEnter \
  --start-fullscreen \
  --kiosk \
  --incognito \
  --noerrdialogs \
  --disable-translate \
  --no-first-run \
  --fast \
  --fast-start \
  --disable-infobars \
  --disable-features=TranslateUI \
  --disk-cache-dir=/dev/null \
  --overscroll-history-navigation=0 \
  --check-for-update-interval=31536000 \
  --disable-pinch &

# Start the kiosk loop to change Chromium tab
while (true)
 do
  xdotool keydown ctrl+Tab; xdotool keyup ctrl+Tab;
  sleep 15
done
```
You normally don't need to specify the screen name or even the resolution, just remember to define the ROTATE_DISPLAY variable to a value (normal, left, rigth, inverted)

And of course, you need to install some packages like 
```bash
sudo apt update && sudo apt install -y x11-apps \
    x11-xserver-utils \
    xinit \
    xinput \
    xorg 
```