balenasolutions/browser #154: Broken since balenaOS 2.115.18

Alexander89, 2023-07-19:
Hallo Balena labs team, 

I've been using the Balena browser for years, and it is excellent!

But since the last couple of updates, it is no longer starting.
When I start my rPI 4 with any newer BalenaOS version than balenaOS 2.114.18+rev3 (this is working), I get this error message:

```
AddScreen/ScreenInit failed for driver 0
```

It is a completely fresh installation. I can reproduce it on any rPI4 I have.

Setup: rPI 4 - 4GB Ram + IIYAMA screen T2251MSC

Container logs:
```
balena logs df
user.max_user_namespaces = 10000
umount: /dev: target is busy.
mount: /sys/kernel/debug: nodev already mounted on /sys/kernel/debug.
Running balena base image entrypoint...
setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
balenaBlocks browser version: 2.4.5
Setting CPU Scaling Governor to 'performance'
Enabling cursor
hostname: No address associated with hostname
xauth:  file /home/chromium/.Xauthority does not exist
xauth: (stdin):1:  bad display name "811ee39:0" in "add" command

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.10.0-10-arm64 aarch64 Debian
Current Operating System: Linux 811ee39 5.15.92-v8 #1 SMP PREEMPT Wed Feb 8 16:47:50 UTC 2023 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=E4:5F:01:A2:3F:3D vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 rootfstype=ext4 rootwait dwc_otg.lpm_enable=0 rootwait vt.global_cursor_default=0 console=null cgroup_enable=memory root=UUID=ba1eadef-ad42-4d5c-bb3d-c9fd3332f94a rootwait
Build Date: 18 December 2021  09:05:36AM
xorg-server 2:1.20.4-1+deb10u4 (https://www.debian.org/support) 
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Jul 18 14:36:48 2023
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) AddScreen/ScreenInit failed for driver 0
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
         at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
xauth: (argv):1:  bad display name "811ee39:0" in "remove" command
```

host dmsg
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.92-v8 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 11.3.0, GNU ld (GNU Binutils) 2.38.20220708) #1 SMP PREEMPT Wed Feb 8 16:47:50 UTC 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[... 423 lines cut ...]
[   48.124880] br-41a9fa3a5c1a: port 1(vethf3200d3) entered disabled state
[   48.127404] device vethf3200d3 left promiscuous mode
[   48.127426] br-41a9fa3a5c1a: port 1(vethf3200d3) entered disabled state
[   48.470931] br-41a9fa3a5c1a: port 1(veth05e943c) entered blocking state
[   48.470958] br-41a9fa3a5c1a: port 1(veth05e943c) entered disabled state
[   48.471317] device veth05e943c entered promiscuous mode
[   48.476132] br-41a9fa3a5c1a: port 1(veth05e943c) entered blocking state
[   48.476160] br-41a9fa3a5c1a: port 1(veth05e943c) entered forwarding state
[   49.052746] br-41a9fa3a5c1a: port 1(veth05e943c) entered disabled state
[   50.734181] eth0: renamed from veth118949c
[   50.761950] IPv6: ADDRCONF(NETDEV_CHANGE): veth05e943c: link becomes ready
[   50.762113] br-41a9fa3a5c1a: port 1(veth05e943c) entered blocking state
[   50.762127] br-41a9fa3a5c1a: port 1(veth05e943c) entered forwarding state
[   56.200377] br-41a9fa3a5c1a: port 1(veth05e943c) entered disabled state
[   56.201038] veth118949c: renamed from eth0
[   56.261693] br-41a9fa3a5c1a: port 1(veth05e943c) entered disabled state
[   56.265060] device veth05e943c left promiscuous mode
[   56.265086] br-41a9fa3a5c1a: port 1(veth05e943c) entered disabled state
[   57.021796] br-41a9fa3a5c1a: port 1(veth52d824b) entered blocking state
[   57.021822] br-41a9fa3a5c1a: port 1(veth52d824b) entered disabled state
[   57.022198] device veth52d824b entered promiscuous mode
[   57.022868] br-41a9fa3a5c1a: port 1(veth52d824b) entered blocking state
```

It would be awesome if someone could help me out.

VirgileMulac, 2023-07-19:
Hello
I have the exact same issue. I've seen it triing to create a new device.
A workaround is to select older version of balena but it is not satisfying
Thanks for support to fix this

VirgileMulac, 2023-08-21:
Hello
is there any update on this topic ? 
How can we know if someone is working on it ?
Thanks

cwit-az, 2023-09-14:
I fixed the issue by simple pushing a new revision with the browser block, seems the image got updated and it's now fixed

phil-d-wilson, 2024-09-04:
Hi @Alexander89 sorry for the wait, I've been distracted by other things. Do you still have the issue with the latest versions of the block?

Phil

Alexander89, 2024-09-04:
No, works just fine. Thx for asking.
I close the issue

closed 2024-09-04 by Alexander89

fabiopedrosa, 2024-10-04:
@phil-d-wilson This still happen when using Balena 6