balenasolutions/browser #163: Raspberry Pi5 gives error: Cannot run in framebuffer mode. Please specify busIDs for all framebuffer devices

matrover, 2023-12-09:
Hi im using this block for many years on pi4 and 3. today i bought pi5's 
made a new fleet and pushed the 2.6.0 version of this block to the pi5 fleet

gives me an error: `Cannot run in framebuffer mode. Please specify busIDs for all framebuffer devices` any idea? 

> 2023-12-09T12:24:35+01:00 Supervisor starting
2023-12-09T12:24:30+01:00  browser  user.max_user_namespaces = 10000
2023-12-09T12:24:30+01:00  browser  umount: /dev: target is busy.
2023-12-09T12:24:31+01:00  browser  Running balena base image entrypoint...
2023-12-09T12:24:32+01:00  browser  setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
2023-12-09T12:24:32+01:00  browser  balenaLabs browser version: 2.6.0
2023-12-09T12:24:32+01:00  browser  Setting CPU Scaling Governor to 'performance'
2023-12-09T12:24:32+01:00  browser  Disabling cursor
2023-12-09T12:24:32+01:00  browser  hostname: No address associated with hostname
2023-12-09T12:24:32+01:00  browser  xauth: (stdin):1:  bad display name "7f72edd:0" in "add" command
2023-12-09T12:24:32+01:00  browser  
2023-12-09T12:24:33+01:00  browser  
2023-12-09T12:24:33+01:00  browser  X.Org X Server 1.20.11
2023-12-09T12:24:33+01:00  browser  X Protocol Version 11, Revision 0
2023-12-09T12:24:33+01:00  browser  Build Operating System: linux Debian
2023-12-09T12:24:33+01:00  browser  Current Operating System: Linux 7f72edd 6.1.54-v8 #1 SMP PREEMPT Wed Sep 27 16:45:31 UTC 2023 aarch64
2023-12-09T12:24:33+01:00  browser  Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:BD:6B:89 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 rootfstype=ext4 rootwait dwc_otg.lpm_enable=0 rootwait vt.global_cursor_default=0 console=null cgroup_enable=memory   root=UUID=ba1eadef-63f9-435a-90cf-44d085bed149
2023-12-09T12:24:33+01:00  browser  Build Date: 25 October 2023  07:47:13AM
2023-12-09T12:24:33+01:00  browser  xorg-server 2:1.20.11-1+deb11u8 (https://www.debian.org/support) 
2023-12-09T12:24:33+01:00  browser  Current version of pixman: 0.40.0
2023-12-09T12:24:33+01:00  browser      Before reporting problems, check http://wiki.x.org
2023-12-09T12:24:33+01:00  browser      to make sure that you have the latest version.
2023-12-09T12:24:33+01:00  browser  Markers: (--) probed, (**) from config file, (==) default setting,
2023-12-09T12:24:33+01:00  browser      (++) from command line, (!!) notice, (II) informational,
2023-12-09T12:24:33+01:00  browser      (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
2023-12-09T12:24:33+01:00  browser  (==) Log file: "/var/log/Xorg.0.log", Time: Sat Dec  9 11:24:33 2023
2023-12-09T12:24:33+01:00  browser  (==) Using system config directory "/usr/share/X11/xorg.conf.d"
2023-12-09T12:24:33+01:00  browser  (EE) 
2023-12-09T12:24:33+01:00  browser  Fatal server error:
2023-12-09T12:24:33+01:00  browser  (EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devices
2023-12-09T12:24:33+01:00  browser  (EE) 
2023-12-09T12:24:33+01:00  browser  (EE) 
2023-12-09T12:24:33+01:00  browser  Please consult the The X.Org Foundation support 
2023-12-09T12:24:33+01:00  browser       at http://wiki.x.org
2023-12-09T12:24:33+01:00  browser   for help. 
2023-12-09T12:24:33+01:00  browser  (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
2023-12-09T12:24:33+01:00  browser  (EE) 
2023-12-09T12:24:33+01:00  browser  (EE) Server terminated with error (1). Closing log file.
2023-12-09T12:24:47+01:00  browser  xinit: giving up
2023-12-09T12:24:47+01:00  browser  xinit: unable to connect to X server: Connection refused
2023-12-09T12:24:47+01:00  browser  xinit: server error
2023-12-09T12:24:47+01:00  browser  xauth: (argv):1:  bad display name "7f72edd:0" in "remove" command
2023-12-09T12:24:47+01:00  browser  Idling...

matrover, 2023-12-18:
Update:  if i delete the “vc4-kms-v3d” from the "Define DT overlays" parameter im getting video output. But without that parameter i cant get the display resolution to 4K

phil-d-wilson, 2023-12-21:
Hey there @matrover 

Thanks for the report. We are aware of the issue, and we are working on a fix: https://github.com/balena-labs-projects/browser/pull/166

I'm not sure of an estimate, but we're looking into it.

matrover, 2023-12-22:
@phil-d-wilson Great news thanks!

mtoman, 2024-01-16:
Should be fixed by https://github.com/balena-labs-projects/browser/commit/92105454eb5b57e21af4439ccd70d95296c9c556

closed 2024-01-16 by mtoman