balenasolutions/browser #167: Building 2.7.0 fails /bin/sh: 1: /usr/src/build/install_chromium: Permission denied

matrover, 2024-01-17:
Hi im trying to build the new 2.7.0 version but it fails. 

Tried for pi4 and pi5 same error using the balena CLI 17.4.11

Gives two errors: 
[browser]  /bin/sh: 1: /usr/src/build/install_chromium: Permission denied
[browser]  The command '/bin/sh -c /usr/src/build/install_chromium "raspberrypi5"' returned a non-zero code: 126

Full log: 

```
PS C:\Users\[..]\Downloads\browser-master (4)\browser-master> balena push webbeheerder/pi5-browser --debug
[debug] new argv=[C:\Program Files\balena-cli\client\bin\node.exe,C:\Program Files\balena-cli\client\bin\run,push,webbeheerder/pi5-browser] length=4
[debug] Deprecation check: 0.00944 days since last npm registry query for next major version release date.
[debug] Will not query the registry again until at least 7 days have passed.
[Debug]   Using build source directory: . 
[Debug]   Pushing to cloud for fleet: webbeheerder/pi5-browser
[debug] Event tracking error: Timeout awaiting 'response' for 0ms
| Packaging the project source...[Debug]   Tarring all non-ignored files...
/ Packaging the project source...[Debug]   docker-compose.yml file found at "C:\Users\[..]\Downloads\browser-master (4)\browser-master"
\ Packaging the project source...[Debug]   Tarring complete in 325 ms
[debug] Connecting to builder at https://builder.balena-cloud.com/v3/build?slug=webbeheerder%2Fpi5-browser&dockerfilePath=&emulated=false&nocache=false&headless=false&isdraft=false
/ Uploading source package to https://builder.balena-cloud.com[debug] received HTTP 200 OK
[debug] handling message: {"type":"metadata","resource":"buildLogId","value":"2855281"}
[debug] handling message: {"message":"\u001b[36m[Info]\u001b[39m     Starting build for pi5-browser, user m15"}
[Info]     Starting build for pi5-browser, user m15
[debug] handling message: {"message":"\u001b[36m[Info]\u001b[39m     Dashboard link: https://dashboard.balena-cloud.com/apps/2101136/devices"}
[Info]     Dashboard link: https://dashboard.balena-cloud.com/apps/2101136/devices
[debug] handling message: {"message":"\u001b[36m[Info]\u001b[39m     Building on 4b2d7d0"}
[Info]     Building on 4b2d7d0
[debug] handling message: {"message":"\u001b[36m[Info]\u001b[39m     Pulling previous images for caching purposes..."}
[Info]     Pulling previous images for caching purposes...
[debug] handling message: {"message":"[=>                                                 ] 2%","replace":true}
[..]
[==================================================>] 100%[debug] handling message: {"type":"metadata","resource":"cursor","value":"erase"}
[debug] handling message: {"message":"\u001b[32m[Success]\u001b[39m  Successfully pulled cache images"}
[Success]  Successfully pulled cache images
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  Step 1/22 : ARG NODEJS_VERSION=\"16.19.1\""}
[browser]  Step 1/22 : ARG NODEJS_VERSION="16.19.1"
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  Step 2/22 : FROM balenalib/raspberrypi5-debian-node:${NODEJS_VERSION}-bookworm-run"}
[browser]  Step 2/22 : FROM balenalib/raspberrypi5-debian-node:${NODEJS_VERSION}-bookworm-run
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m   ---> b5c118e01c5b"}
[browser]   ---> b5c118e01c5b
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  Step 3/22 : COPY ./build /usr/src/build"}
[browser]  Step 3/22 : COPY ./build /usr/src/build
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m   ---> 8a51c7b76a3c"}
[browser]   ---> 8a51c7b76a3c
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  Step 4/22 : RUN /usr/src/build/install_chromium \"raspberrypi5\""}
[browser]  Step 4/22 : RUN /usr/src/build/install_chromium "raspberrypi5"
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m   ---> Running in 43f7f56f7937"}
[browser]   ---> Running in 43f7f56f7937
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  \u001b[91m/bin/sh: 1: /usr/src/build/install_chromium: Permission denied"}
[browser]  /bin/sh: 1: /usr/src/build/install_chromium: Permission denied
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  \u001b[0m"}
[browser]
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  Removing intermediate container 43f7f56f7937"}
[browser]  Removing intermediate container 43f7f56f7937
[debug] handling message: {"message":"\u001b[34m[browser]\u001b[39m  \u001b[31mThe command '/bin/sh -c /usr/src/build/install_chromium \"raspberrypi5\"' returned a non-zero code: 126\u001b[39m"}
[browser]  The command '/bin/sh -c /usr/src/build/install_chromium "raspberrypi5"' returned a non-zero code: 126
[debug] handling message: {"message":"\u001b[36m[Info]\u001b[39m     Uploading images"}
[Info]     Uploading images
[debug] handling message: {"message":"[===>                                               ] 6%","replace":true}
[..]
[==================================================>] 100%[debug] handling message: {"type":"metadata","resource":"cursor","value":"erase"}
[debug] handling message: {"message":"\u001b[32m[Success]\u001b[39m  Successfully uploaded images"}
[Success]  Successfully uploaded images
[debug] handling message: {"message":"\u001b[31m[Error]\u001b[39m    Some services failed to build:","isError":true}
[Error]    Some services failed to build:
[debug] handling message: {"message":"\u001b[31m[Error]\u001b[39m      Service: browser","isError":true}
[Error]      Service: browser
[debug] handling message: {"message":"\u001b[31m[Error]\u001b[39m        Error: The command '/bin/sh -c /usr/src/build/install_chromium \"raspberrypi5\"' returned a non-zero code: 126","isError":true}
[Error]        Error: The command '/bin/sh -c /usr/src/build/install_chromium "raspberrypi5"' returned a non-zero code: 126
[debug] handling message: {"message":"\u001b[36m[Info]\u001b[39m     Built on 4b2d7d0"}
[Info]     Built on 4b2d7d0
[debug] handling message: {"message":"\u001b[31m[Error]\u001b[39m    Not deploying release.","isError":true}
[Error]    Not deploying release.
Remote build failed

RemoteBuildFailedError: Remote build failed
    at handleRemoteBuildStream (C:\Program Files\balena-cli\client\build\utils\remote-build.js:83:15)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async Object.startRemoteBuild (C:\Program Files\balena-cli\client\build\utils\remote-build.js:64:13)
    at async PushCmd.pushToCloud (C:\Program Files\balena-cli\client\build\commands\push\index.js:78:27)
    at async PushCmd.run (C:\Program Files\balena-cli\client\build\commands\push\index.js:32:17)
    at async PushCmd._run (C:\Program Files\balena-cli\client\node_modules\@oclif\core\lib\command.js:304:22)
    at async Config.runCommand (C:\Program Files\balena-cli\client\node_modules\@oclif\core\lib\config\config.js:417:25)
    at async run (C:\Program Files\balena-cli\client\node_modules\@oclif\core\lib\main.js:85:16)
    at async C:\Program Files\balena-cli\client\build\app.js:80:13
    at async Promise.all (index 2)
    at async oclifRun (C:\Program Files\balena-cli\client\build\app.js:98:5)
    at async Object.run (C:\Program Files\balena-cli\client\build\app.js:111:9)
    at async run (C:\Program Files\balena-cli\client\bin\run:18:2)
```

phil-d-wilson, 2024-09-04:
Hi @matrover - this will be fixed today. I'm just merging a PR with a fix

phil-d-wilson, 2024-09-04:
All sorted @matrover

closed 2024-09-04 by phil-d-wilson