balenasolutions/browser #169: Mouse pointer not showing up

benoit74, 2024-03-29:
Mouse pointer is not showing up on Chromium window. 

Double clicking the mouse left button is selecting / unselecting text, so mouse is more or less operating. 

However it seems that trackpad is not making the pointer move.

I tried to plug an external (RF USB dongle) CHERRY mousse. This time it looks like the pointer is moving, because when double clicking the text selected is not always the same after moving the mouse. But still no pointer.

Keyboard is operating properly.

Current device is an amd64 architecture, but I previously experienced the same issue on a aarch64 platform.

Logs:
```
mount: /tmp/tmpmount/console: special device /dev/console does not exist.
       dmesg(1) may have more information after failed mount system call.
umount: /dev: target is busy.
user.max_user_namespaces = 10000
mount: /tmp/tmpmount/console: wrong fs type, bad option, bad superblock on /dev/console, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
umount: /dev: target is busy.
Running balena base image entrypoint...
setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
balenaLabs browser version: 2.7.0
Setting CPU Scaling Governor to 'performance'
Disabling cursor

X.Org X Server 1.21.1.7
X Protocol Version 11, Revision 0
Current Operating System: Linux aaf803de5c27 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.0-18-amd64 root=UUID=00e4f10d-2db3-4e2c-ab36-b6a7e21cddea ro quiet
xorg-server 2:21.1.7-3+deb12u4 (https://www.debian.org/support)
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Fri Mar 29 07:31:27 2024
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
--- List Input Devices ---
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Cherry GmbH  CHERRY Wireless Device Mouse	id=11	[slave  pointer  (2)]
⎜   ↳ Cherry GmbH  CHERRY Wireless Device Consumer Control	id=12	[slave  pointer  (2)]
⎜   ↳ CyPS/2 Cypress Trackpad                 	id=15	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Cherry GmbH  CHERRY Wireless Device     	id=10	[slave  keyboard (3)]
    ↳ Cherry GmbH  CHERRY Wireless Device System Control	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
----- End of List --------
Detecting window size from framebuffer
Window size detected as 1366,768
Installed browser version: Chromium 120.0.6099.216 built on Debian 12.4, running on Debian 12.2
Browser API running on port: 5011
LAUNCH_URL environment variable not set.
Looking for local HTTP/S services.
No service found on port 80
No service found on port 443
No service found on port 8080
Displaying default HTML page
Disabling GPU
Disabling KIOSK mode
Starting Chromium with flags: --disable-features=TranslateUI,--disable-component-extensions-with-background-pages,--disable-background-networking,--disable-sync,--metrics-recording-only,--disable-default-apps,--no-default-browser-check,--no-first-run,--disable-backgrounding-occluded-windows,--disable-renderer-backgrounding,--disable-background-timer-throttling,--force-fieldtrials=*BackgroundTracing/default/,--window-size=1366,768,--window-position=0,0,--autoplay-policy=no-user-gesture-required,--noerrdialogs,--disable-session-crashed-bubble,--check-for-update-interval=31536000,--disable-dev-shm-usage,--disable-gpu
Displaying URL: file:///home/chromium/index.html
Chromium remote debugging tools running on port: 35173
```

Any idea?

phil-d-wilson, 2024-09-04:
Hi there,

Can I check if you had the SHOW_CURSOR environment variable set to `1`? By default we hide the cursor since most users will have the block in kiosk mode for digital signage applications.

Phil

benoit74, 2024-09-05:
Shame on me ... I didn't customized this variable at all, so you've probably the proper explanation ... Sorry for that, I'll close the issue, and reopen should there be another problem, but it now seem pretty obvious there is no issue besides the person sitting in front of my screen and keyboard ^^

closed 2024-09-05 by benoit74