balenasolutions/browser #173: Supported touch displays for Rpi

iBobik, 2024-06-30:
Hello,
I’ve just bought [Waveshare DPI display](https://www.waveshare.com/wiki/50H-800480-IPS) but it requires some proprietary driver only for Raspbian and installed only thru Windows, so I found the hard way not all displays developed for Rpi will work plug&play.

What displays are supported by Balena OS and this image

 I haven't found info about it in Balena docs. It would be nice to also link it from there.

phil-d-wilson, 2024-09-04:
@alanb128 can you help here at all?

alanb128, 2024-09-04:
This issue was addressed on our Forums last month [here](https://forums.balena.io/t/waveshare-dsi-lcd-touchscreen-support/369634/2). There's more info, context, and conversation in that thread, so I suggest anyone looking for similar help to check that out first.

closed 2024-09-04 by phil-d-wilson