balenasolutions/browser #181: Pull access denied.

ColinTravis, 2024-11-12:
Upon trying to `balena push` this repo with the following `docker-compose.yml` file:

```
version: '2'

volumes:
  settings:                          # Only required if using PERSISTENT flag (see below)

services:

  browser:
    image: bh.cr/balenablocks/browser-aarch64 # where <arch> is one of aarch64, arm32 or amd64
    privileged: true # required for UDEV to find plugged in peripherals such as a USB mouse
    ports:
        - '5011' # management API (optional)
        - '35173' # Chromium debugging port (optional)
    volumes:
      - 'settings:/data' # Only required if using PERSISTENT flag (see below)

```

I get this error:
```
[Error]    Some services failed to build:
[Error]      Service: browser
[Error]        Error: (HTTP code 404) unexpected - pull access denied for bh.cr/balenablocks/browser-aarch64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
[Info]     Built on 1d00bb3
[Error]    Not deploying release.
Remote build failed
```

I am currently logged into both Balena CLI and Docker as well.

phil-d-wilson, 2024-11-12:
Hi there @ColinTravis ,

Try changing the image reference to:
`bh.cr/balenalabs/browser-aarch64`

ColinTravis, 2024-11-12:
That did the trick, I think I pulled that image URL from an old article. Thanks.

closed 2024-11-12 by ColinTravis